prusa3d/PrusaSlicer #6174: filling up voids while slicing

Castaneda77, 2021-03-04:
### Version  sorry, I aborted my previous message unvoluntarily! (not used to this process)
_Version of PrusaSlicer used goes here_2.3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version Windows 10
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) not relevant
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_While slicing my file, a void part becomes filled
* _Steps needed to reproduce the problem_
* Load the stl file Ouvre bouteilles 3 bottom v2.stl, rotate it 180 ° axis Y, (check that it is void, slice it, check that it is filled!
[image]
[image]
[Ouvre bouteilles 3 bottom v2.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/6085527/Ouvre.bouteilles.3.bottom.v2.stl.zip)

Thats all, hope that you can solve this issue

Castaneda7777@gmail.com

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-03-04:
The model is seriously broken. I was not able to repair it usefully with any standard SW. It also wasn't sliced correctly with any slicer I tried (Cura, Simplify3D).

Closing. Invalid model.

closed 2021-03-04 by rtyr

Castaneda77, 2021-03-04:
> Répondre à : prusa3d/PrusaSlicer <reply@reply.github.com>

Thanks, I didn’t notice my model was bad, ass the slicer displays it correctly, sorry ;-(