prusa3d/PrusaSlicer #6175: Exporting an STL in Prusa slicer to G-code loses part of the print. When printing a small glass door the whole door is available in STL but the sides disappear in the G-code file.

BungalowJIm, 2021-03-04:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-03-04:
We need more than this. Please provide your 3MF project file (you have to zip it first).

BungalowJIm, 2021-03-05:
Hi,
 Here is a zip with the files in, STL, G-Code and 3MF. As my Prusa mini requires G-code I import as an STL file and then export as a G-Code after adjusting further layers to be printed at a lower temperature. This solved my problem with stringing. However, when I convert this door for a model greenhouse, it appears to lose the sides and top of the door. Any ideas?.
Jim
To do is to be. Confuscious (500 B.C.) To be is to do. Gerard 't Hooft (1943) Doo be doo be doo. Frank Sinatra. (1984)

rtyr, 2021-03-05:
It is not possible to add files from e-mail. You have to upload it directly through github web interface.

BungalowJIm, 2021-03-05:
I am getting two messages. One says 'You can't comment at this time'. The other says 'we cant process that file'. 
Modern technology really is a bind at times.

BungalowJIm, 2021-03-05:
[door fault.zip](https://github.com/prusa3d/PrusaSlicer/files/6091143/door.fault.zip)

BungalowJIm, 2021-03-05:
Is that what you require? I import an STL file export it as a G-code so that my Prusa min can read it. The G-code has the sides and top of the door missing.

rtyr, 2021-03-05:
The missing parts are too thin to be printed with set extrusion width.

It would be better to print it with smaller nozzle, but you should be able to print it with lower extrusion width values for perimeters/external perimeters (something like 0.35mm).

BungalowJIm, 2021-03-05:
Thanks. I’m new to 3D printing. I’ll give it a try. I take it I can change the setting in the app. 
Jim

rtyr, 2021-03-05:
Yes, you can set it on advanced tab. Closing.

closed 2021-03-05 by rtyr