prusa3d/PrusaSlicer #6179: Detect Bridge Perimeters doesn´t work

hytakara, 2021-03-05:
### Version
Prusa slicer 2.3.0

### Operating system type + version
Windows 10 Home Single Language - version 20H2

### 3D printer brand / version + firmware version (if known)
Anycubic Chiron version 1.3.0

### Behavior
* Import a STL file
* Go to Printing  Setings menu -> Layers and perimeters:
* Turn off detect bridge perimeters
* Go to Printing Settings menu -> Infill :
* Set 30% to Fill density
* Set grid to Fill patern
* Set 1 to Combine infill every:
* Go to Plater meu and Slice model
* All solid layers must have the same printing pattern
* When printing suspended layer Slicer change printing pattern to bridge pattern
 * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[screen shot.zip](https://github.com/prusa3d/PrusaSlicer/files/6092977/screen.shot.zip)

This is an issue query.

#### Project File (.3MF) where problem occurs
[model test.zip](https://github.com/prusa3d/PrusaSlicer/files/6092984/model.test.zip)

hytakara, 2021-03-05:
The option (detect bridge perimeters) does not work.
Even deactivating the option, the program continues to detect bridge perimeters creating a different layer as if it were a bridge where it shouldn't have been.
This forces a change in the printing pattern of the layer identified as a bridge causing a structural failure in the part, the most fragile layer even when we choose to use support to avoid the problem.

rtyr, 2021-03-05:
I don't see any bridge perimeter with "detect bridge perimeters" disabled. There are only two layers with bridge infill (not perimeters), both on expected correct places.

rtyr, 2021-03-05:
Set 4 perimeters for layers 12-12.30 (or for whole print), if you want to avoid bridge infill there.

rtyr, 2021-03-08:
Answered. Closing.

closed 2021-03-08 by rtyr