prusa3d/PrusaSlicer #6180: Plater not visible / freezed on external monitor

theolderbear, 2021-03-05:
Just updated my macbook to big.

### Version
2.3.0+

### Operating system type + version
macOS Big Sur version 11.2.2
MacBook Pro (15-inch, 2018)

PrusaSlicer
Version:   2.3.0+
Build:     PrusaSlicer-2.3.0+-202101111304

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.16 (Build 20D80)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 ATI-4.2.15
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 555X OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix

### 3D printer brand / version + firmware version (if known)
Prusa is mks3+

### Behavior
1.
- Launch Prusa slicer 
- Prusa slicer start on external displaylink monitor (Docking Station dell d6000)
- Plater is not visible full view is gray
[image]
- is not possible to open or view any 3d object or previous saved project
- all other functions seems to work correctly

2.
- Launch Prusa slicer 
- Prusa slicer start on macbook built in monitor
- Plater is visible
- is possible to open and view any 3d object or previous saved project
- move prusa slicer window on external display
- plater freezes, no operation are possible
- move prusa slicer window back on built in monitor
- Plater is operating all functions works correctly

fe60, 2021-03-06:
As you didn't mention it explicitly:
The scenarios described above did work on your MacBook with Mac OS 10.15 (Catalina) and you face the issues since the upgrade to Mac OS 11.0 (Big Sur). Is this correct?

theolderbear, 2021-03-06:
Yes is correct. Just after upgrade one of my monitor was totally not working. I had to install this https://www.displaylink.com/downloads/macos to have it back working.
I did not have still time to properly test everything but prusa slicer is the only software (and the first I used) that has issue, blender for example work fine. 

I can add that:

- start pura in built in monitor
- move prusa window in the external monitor
- try to move or rotate the plater
- nothings happens
- move back prusa slicer window in  built in monitor
- now movement and rotation are applied/visible

theolderbear, 2021-03-09:
Hi, last update of mac os Big Sur 11.2.3 has fixed the issue.

closed 2021-03-09 by theolderbear