prusa3d/PrusaSlicer #6184: Top infill spreads/leaks into edge of non-top layer on separate nearby object

tg73, 2021-03-06:
### Version
2.3.0 (release) Win64

### Operating system type + version
Win 10

### Behavior
See image below. The objects showing mostly purple solid infill are 0.65mm high. The objects showing mostly red top infill are 0.5mm high. We are looking at the layer at 0.5mm. There should be no top infill in the 0.65mm high objects in this layer. However, it's as if the top infill is "leaking" or "spreading" from the nearby 0.5mm objects into the nearby edges of the 0.65mm high objects. This happens only in close proximity to the top infill regions of the 0.5mm objects. Very odd indeed! I can't publicly share the source 3mf for this, but can share direct with a Prusa dev upon request. I've tried to create a simple test case, but could not repeat the issue.

[image]

neophyl, 2021-03-06:
If they were separate objects then it wouldn't do that.  As they are one object still (even though not physically touching)  for the purposes of slicing then its sliced as a single object.  That means it treats that layer the same so the top infill grows under the surrounding areas.  If you added more separation between the parts it would also not do that as the grow factor wouldn't reach the second.

I wouldn't class this as a bug, more a consequence of of its designed to work.

tg73, 2021-03-06:
Well, they're clearly separate parts in terms of how PS sees them (as per treeview), and from a purely geometric perspective, they do not touch or intersect. PS correctly produces external perimeter on both objects. I don't see how top infill crossing over external perimeter into another disconnected part is desirable in any circumstance (although please do tell me if you see how it could be desirable). As such, while it might be an explicable artefact of the current slicing logic, I would see it as incorrect behaviour. In my real-world use case, I need to pack the objects snugly to optimise print time (I need to print thousands of small think objects). I'm also using multiple materials, so I have to prepare a pre-arranged bed of objects.

neophyl, 2021-03-06:
The treeview was my point, they are listed as one object.  Not 2.  There's a distinct difference between a Part and an Object.  Objects are sliced as one logical entity.  It does matter if there is physical separation or they are touching.   To illustrate what I mean create several boxes all as one Object but with several Parts.  Then make sure they use infill, use something like concentric.   The infill pattern is applied to the object as a whole.  If you make them separate Objects then the infill pattern is applied to each individually.  
I agree for your use case that you need a solution but I would argue that this is a feature change request and not a bug.  The very language you use of 'packing objects', except with how you have them they are not objects, you are packing parts.

Is the new option to set the arrange distance not useful in your case ?  Or are you some some really dense packing ?

tg73, 2021-03-06:
Thanks for your thoughts. 

However, my core question remains unanswered: when does it serve a useful purpose for top infill to expand beyond its bounding external perimeter?

My argument is that there is no good reason for this to happen, so it should not. This is not a new feature, it's a correction to the behaviour of an existing feature. Note that I do understand why top infill expands *within* its bounding external perimeter - so that the infill reveals a clean transition behind the layer above (when there is one), eg as the background to the extruded text on the classic [PRUSA text sample object](https://github.com/prusa3d/Prusa3D-Sample-Objects/blob/master/1.75mm/Prusa.stl).

I my real world use case, my parts are software-generated, and I generate a 3mf for each complete plate. A complete print run will comprise over 2000 parts, spanning several dozen plates. Keeping the parts as close as possible (based on nozzle footprint) meaningfully increases the number of parts I can get on one plate, and so reduces the total number of plates per print run. I use my own code for plate arrangement which has some special purpose logic. I also generate a sprue that connects columns of parts together to make for easier management in post production. The sprue is exactly two extrusions wide and one layer high, and is designed to lead to PS creating a single continuous outer perimeter that spans all the parts and the sprue. There must not be boundaries between the sprue and the parts otherwise the sprue will break off too easily. In this scenario, I would argue that it is entirely correct to describe all the parts joined by a sprue as one top-level object.

bubnikv, 2021-03-08:
> However, my core question remains unanswered: when does it serve a useful
purpose for top infill to expand beyond its bounding external perimeter?

It is an unintended behavior. The algorithm expands the top surfaces of the
whole layer all at once and then it trims them with the islands. Thus some
of the expanded regions end up in the wrong islands.

bubnikv, 2021-03-08:
What strikes me is that after 5 years involvement with the software there
are still bugs that are new to me.

closed 2026-09-29 by prusaslicer-bot[bot]