prusa3d/PrusaSlicer #6188: Support generated on the top

eiffel31, 2021-03-06:
### Version
Version 2.3.0

### Behavior
As shown on the screen capture, some support is generated on the top in some cases.

#### Project File (.3MF) where problem occurs
[image]
[supports_on_top.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6095967/supports_on_top.3mf.zip)

HiLivin, 2021-09-19:
The issue seems resolved in 2.4.0-alpha1 with "Snug" support style chosen.
The "Grid" style still exhibits the same artifacts as before.

bubnikv, 2021-10-05:
Even the grid supports are better, they leak less

bubnikv, 2021-10-05:
With snug supports the leakage is gone. I think we can close this issue.

bubnikv, 2021-10-05:
[image]

closed 2021-10-05 by bubnikv