prusa3d/PrusaSlicer #6189: Custom Mouse Sensitivity For 3D-view

peopplank, 2021-03-06:
**Version** 2.3.0+linux-x64
PC POP_OS / Windows 10

I find the default mouse sensitivity for the 3D-view rather high in PrusaSlicer. This is in Windows as well as in linux. And I know I can change the sens of the OS but it would be handy if it was an app-specific setting. So I was thinking maybe add some slider in the settings to change the sensitivity for panning around.

mihai-stanimir, 2021-03-06:
Similar issue when using the Space Mouse (3D mouse), the speed in this case is very low.

mihneaCR, 2021-03-07:
Would love that. I am also facing the same issue.

foreachthing, 2021-03-07:
Try [CTRL]+[M]:
[image]

mihai-stanimir, 2021-03-08:
Thanks @foreachthing!
I had to "enable support for legacy 3DConnexion devices" for the Ctrl+M menu to pop up although my SpaceMouse Pro was working with the app. Then I could disable the checkbox. Did work, but is a bit weird that the menu won't show up without that checkbox. My SpaceMouse Pro is relatively new, so not "legacy". Anyways, works.

bubnikv, 2021-03-08:
> My SpaceMouse Pro is relatively new, so not "legacy".

You have to complain at 3DConnexion.

foreachthing, 2021-03-08:
> You have to complain at 3DConnexion.

Yeah, my SpacePilot has become legacy as well ... they just want you to buy new devices ever 3 years or so. This is really annoying!! 😠

Phil1988, 2021-03-11:
+1 for this.

Especially with larger printers, the sensitivity is too high.

bubnikv, 2021-09-02:
Enable "legacy 3D connexion" support with the last checkbox

[image]

and press Ctrl-M in the 3D scene.

You can thank to 3DConnexion for they not so stellar support.

dahuafschmied, 2021-09-02:
> Enable "legacy 3D connexion" support with the last checkbox
> 
> 
> and press Ctrl-M in the 3D scene.
> 
> You can thank to 3DConnexion for they not so stellar support.

hi @bubnikv 
i know all this stuff and the problems with the support from 3Dconnexion. thanks for keeping the 3d mouse supported in PrusaSlicer!!!

till 2.4 i could change the speed in PrusaSlicer with command+m (MAC!!, your screenshot is from windows)
even if command+m does not work i could change the settings in the .ini file.

but since 2.4 i can not access the dialog (legacy enabled) and the entries in the .ini files are gone.
so there is no workaround for the problem anymore.

the simplest solution would be to make just one speed value in the settings (just one for all axis together) from 1 to 10. and all problems are solved for me. and perhaps for many other users too.

i hope you can make this changes and close all this long lasting tickets for 3D Mouses.

dahuafschmied, 2021-09-02:
Shift+Command+M on MAC
thank you for the hint!

bubnikv, 2021-09-22:
@dahuafschmied
> Just to remind Prusa at this very annoying topic for all of us.

Again, please do me a favor and complain at 3DConnexion.

I believe the topic has been explained.

https://www.help.prusa3d.com/en/article/3dconnexion-devices_1511

It is even in the manual.

closed 2021-09-22 by bubnikv

fosterthree, 2023-08-10:
Seconding @megabitsenmzq about orbit hypersensitivity on Mac with standard mouse / trackpad, with similar results to their video.

AudreyAP, 2023-12-22:
Has control-m been removed? It does not function in 2.7.1.

cjameshuff, 2024-01-31:
I am also experiencing this. Just an ordinary mouse, other applications behave well (FreeCAD and OpenSCAD being particular examples of similar interfaces), but using PrusaSlicer is a chore because natural mouse movements send the display spinning.

maxzera, 2024-02-03:
Seconding @cjameshuff and @AudreyAP. In the meantime, is it possible to "hard-code" a sensitivity number on a configuration file or something?