prusa3d/PrusaSlicer #6193: Repair with Netfabb does not work on builds after 2.3.0 release (Windows 10)

jmd73, 2021-03-07:
### Version
All commits AFTER 55c698feeff9b204b01f698aac413679f4b913c2 (all the way to 1efa9a06f67d7796bda21fedcff7bdabadbd66a5)

Release version 2.3.0-win64 works fine.

### Operating system type + version
Windows 10 Pro 64bit.

### 3D printer brand / version + firmware version (if known)
MK3Sand MK2.5S

### Behavior
Trying to repair a model by the Netfabb service (by right-clicking the model) does not work.
Step 1: Open PrusaSlicer
Step 2: Right click on empty plater and select Add Shape->Box.
Step 3: Right click on box and select Repair by Netfabb service,
Step 5:  
[image]

* _Expected Results_
That the repair succeeds
* _Actual Results_
Failed repair
_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
This zip contains a working and not working temporary 3mf files from the export.
The working one can be opened by Windows 3d builder, but the not working fails.
[temp_3mfs.zip](https://github.com/prusa3d/PrusaSlicer/files/6097498/temp_3mfs.zip)

bubnikv, 2021-03-08:
Thanks, good catch. I will look into it.

bubnikv, 2021-03-08:
Fixed with 589d2be442d93e019c7fc8ccf26fa15d6f61be5d

    This is more a workaround than a fix: Windows 10 3D model fixing API refuses
    to load a zip64 encoded 3MF. We need to get in touch with Microsoft on that
    issue, for now the 3MFs generated for the Windows 10 3D model fixing API
    will be limited to 4GB. Saving a bigger 3MF will fail.

closed 2021-03-08 by bubnikv