prusa3d/PrusaSlicer #6194: Faulty gcode generation with hex grid

strayr, 2021-03-07:
### Version
2.3.0+win64

### Operating system type + version
Win 10 Home 20H2 build 19042.804

### 3D printer brand / version + firmware version (if known)
Not relevant, is g-coge generation bug,
verified it's NOT a gcode display bug by printing on an Ender3 4.2.2Melzi board, Vanilla MArlin 2.0.6.1 bugfix

### Behavior
* _Incorrect Gcode generation_
Attatched STL generates incorrect gcode on one side of hex grid. STL producing faulty G-code does not appear to be of inconsistent shape, looks fine in F360 and converting to BREP and exporting as a new mesh generates the same issue.

* _Steps needed to reproduce the problem_
Slice model, have tried Prusa MK3S & MK3S+ and Ender3v2 printers likely printer independent, appears independent of filament, reproducible on all system presets, and pn MK3 and Ender 3 quality presets from 0.24 and finer. Assuming only dependency here is sufficient layer height resolution. 

[BIT_bio_genesis.stl_03.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/6097817/BIT_bio_genesis.stl_03.stl.zip)

* _Expected Results_
Uniform layer count for hex grid, Cura generates consistent thickness. hex grid so I doubt it is the STL is at fault. Should look like this at every edge.
[image]

* _Actual Results_

Note the hex grid is only one layer thick on one edge. 
[image]

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
Reproduced here using only system presets
[BIT_bio_genesis.stl_03-bugreport.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6097815/BIT_bio_genesis.stl_03-bugreport.3mf.zip)

strayr, 2021-03-07:
An update: I have redrawn the original mesh in F360 with a deeper grid because i felt the original was flimsy anyway. I'm still getting the slicer failure in prusaslicer. I'm wondering if it's a quirk of the geometry.

It's not happening in superslicer 2.3.55.

[03 Thick Mesh Top.zip](https://github.com/prusa3d/PrusaSlicer/files/6097861/03.Thick.Mesh.Top.zip)

rtyr, 2021-03-07:
The missing parts of the model are too thin for default extrusion width. If you look closely, you can see that the walls are visibly thinner in that problematic area (I guess it was not intentional). Most of the default profiles use smaller extrusion width for first layer.

You can either fix the model or just use smaller extrusion width for this print.

[image]

rtyr, 2021-03-07:
BTW it would be probably better to make the bottom of the box solid and then slice it with 0 top/bottom layers + honeycomb infill :-).

strayr, 2021-03-08:
> The missing parts of the model are too thin for default extrusion width. If you look closely, you can see that the walls are visibly thinner in that problematic area (I guess it was not intentional). Most of the default profiles use smaller extrusion width for first layer.
> 
I couldn't even see that trying to figure it out in F360. It's not my model, and was originally OpenScad. I've rebuilt it, but I was looking at it for Z thickness being wrong and it didn't occour to me that it was a X-Y thickness. Thanks for the help and sorry for wasting your time.

rtyr, 2021-03-08:
No problem. Closing.

closed 2021-03-08 by rtyr