prusa3d/PrusaSlicer #6196: Snap can't open home direcotry

ajbogh, 2021-03-08:
### Version
2.3.0

### Operating system type + version
Ubuntu 20.04.2

### Behavior
The Snap version of prusa-slicer can no longer open the home directory or any file under it. Apparmor errors are present when run in the terminal.

* _Steps needed to reproduce the problem_
  * Install prusa-slicer snap - `sudo snap install prusa-slicer`
  * Open prusa-slicer from terminal `prusa-slicer`
  * Open file from home directory
  * Review any errors in terminal
* _Expected Results_
  * Files should be visible and should open without errors
* _Actual Results_

```
(prusa-slicer:7689): dconf-CRITICAL **: 01:38:44.632: unable to create file '/run/user/1000/snap.prusa-slicer/dconf/user': Permission denied.  dconf will not work properly.

(prusa-slicer:7689): dconf-WARNING **: 01:38:44.633: failed to commit changes to dconf: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.117" (uid=1000 pid=7689 comm="/snap/prusa-slicer/24/usr/local/bin/prusa-slicer " label="snap.prusa-slicer.prusa-slicer (enforce)") interface="ca.desrt.dconf.Writer" member="Change" error name="(unset)" requested_reply="0" destination="ca.desrt.dconf" (uid=1000 pid=4132 comm="/usr/libexec/dconf-service " label="unconfined")
```  

[image]

bubnikv, 2021-03-08:
I don't know where you got your snap from. Maybe this is the source? 
Snap package #4565

It would be great, if the snap contributor chimed in on how the sandboxing of the snap packages works. Maybe the sandboxing privileges are not set correctly for PrusaSlicer?

ajbogh, 2021-03-08:
@bubnikv The snap package is from the official Ubuntu repos. Simply `snap install prusa-slicer` is what I did.

From my limited research, it appears that maybe the privileges aren't defined correctly in the snap yaml file. I don't know how to fix it though, and I haven't found anything that I can do on my own system to fix it either.

ajbogh, 2021-03-12:
It looks like this got fixed. At least for me it's working again.

closed 2021-03-12 by ajbogh

ivocavalcante, 2021-03-16:
Hi, sorry for coming in so late...

I maintain this package; it's not an official release from Prusa team, as stated on package description (although I volunteer to maintain it here, if they decide on this   :-)  ).

I've not changed anything on the package since 2.3.0 official release. Next step would be trying to compile using the new GNOME 3 extension for core20, but it's not done yet.

The problem you had should have been a "hiccup" on Snap profiles, or some change you might have done to your home folder (like symlinking to another place, which isn't allowed by sandboxing, as @bubnikv stated).

Anyway, I'm available in case you need some help.

--
Ivo