prusa3d/PrusaSlicer #6198: PrusaSlicer 2.3.0 continuously prompts to update to 2.3.0

WimVDK, 2021-03-08:
### Version
_Version of PrusaSlicer used goes here_
2.3.0+win4

### Operating system type + version
Win 10 Pro x64
### 3D printer brand / version + firmware version (if known)
Not relevant
### Behavior
Prusaslicer continuously prompts to update from 2.3.0 TO 2.3.0

[image]

When i then click modify or repair. The "update" always fails, (yet PS continuous to work fine) probably because there's nothing to update.

bubnikv, 2021-03-09:
Do you mean the line at the bottom?

[image]

It is just an information on where the latest updates are available, it does not mean that there is a newer version available. It may be confusing.

rtyr, 2021-03-09:
I suppose it is something similar to https://github.com/prusa3d/PrusaSlicer/issues/5856. Can you share the screenshot with update prompt?

[image]

WimVDK, 2021-03-09:
Not similar, it IS that one. I didn't find it in search. I guess there hasn't been a resolution yet? I'd rather not disable update check entirely.

WimVDK, 2021-03-09:
I'm not entirely sure, been a week or 2 since I installed PS and it finally annoyed me enough to report it. I think I picked single user as it's just me at home. Any way to check for sure?

Janys, 2021-03-09:
Yes. Please right-click the shortcut on the desktop and select Open File Location. If PrusaSlicer is installed in C:\Program Files, then it's installed for all users (as administrator). If it's in C:\Users\Appdata, then it's a single user only.

Janys, 2021-04-19:
Hello, a new version of PrusaSlicer / installer has been released today. Does the update to 2.3.1 work correctly?

Also, if the issue persists, can you please go to C:\Users\Public\Public Documents\Prusa3Dupdater\updater\ and delete the file update.aiu and then run the updater again?

WimVDK, 2021-04-20:
I was able to update normally to 2.3.1. Removed the unwanted prusa printer drivers and install finished normally. After install was prompted for a config update (which added my CR10 Max), that one finished normally as well.

Janys, 2021-04-20:
Thanks!

@rtyr can you please close this issue?

closed 2021-04-20 by rtyr

dirrgang, 2023-10-15:
I have the same issue, been having it for quite a while. PrusaSlicer always wants to update after starting, even when running at the current version.

Removed and reinstalled it just now and the same result. Running version 2.6.1.