prusa3d/PrusaSlicer #6202: PrusaSlicer crashes before completing slicing

staffert, 2021-03-10:
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 12,763MB
OpenGL installation
GL version:   4.0.0 - Build 10.18.10.5161
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 4000
GLSL version: 4.00 - Build 10.18.10.5161

### 3D printer brand / version + firmware version (if known)
not applicable

### Behavior
* Slicer software consumes 100% CPU resources, then crashes and exits before completing the slicing.
* 
[main helmet mandalorian FREE Personal Version from do3d.zip](https://github.com/prusa3d/PrusaSlicer/files/6112933/main.helmet.mandalorian.FREE.Personal.Version.from.do3d.zip)
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2021-03-10:
I cannot reproduce it on Linux. Could you run it from command line with `--loglevel=5` argument and post the end of the output to at least give a clue? You can also try to run with `--sw-renderer` to find out if it's hardware acceleration related. The model is quite complex, do you have enough GPU memory?

staffert, 2021-03-10:
admittedly, it is an older laptop.....but I will try to do as you ask shortly. 
From a geometric point of view I have run some pretty complex models before this one, and quite frankly, this was one of the simpler ones, although the triangle count may be high.....hmmm....

For now, I can offer this:

OS Name	Microsoft Windows 10 Home
Version	10.0.19042 Build 19042
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	DAD-LAPTOP
System Manufacturer	Acer
System Model	Aspire V7-581P
System Type	x64-based PC
System SKU	Aspire V7-581P_0798_2_10
Processor	Intel(R) Core(TM) i5-3337U CPU @ 1.80GHz, 1801 Mhz, 2 Core(s), 4 Logical Processor(s)
BIOS Version/Date	Insyde Corp. V2.10, 2013-10-22
SMBIOS Version	2.7
Embedded Controller Version	0.00
BIOS Mode	UEFI
BaseBoard Manufacturer	Acer
BaseBoard Product	Dazzle_CX
BaseBoard Version	Type2 - A01 Board Version
Platform Role	Mobile
Secure Boot State	Off
PCR7 Configuration	Binding Not Possible
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume2
Locale	Canada
Hardware Abstraction Layer	Version = "10.0.19041.844"
User Name	DAD-LAPTOP\staffert
Time Zone	Eastern Standard Time
Installed Physical Memory (RAM)	12.0 GB
Total Physical Memory	11.9 GB
Available Physical Memory	9.59 GB
Total Virtual Memory	14.2 GB
Available Virtual Memory	11.2 GB
Page File Space	2.36 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	Off
Virtualization-based security	Not enabled
Device Encryption Support	Reasons for failed automatic device encryption: TPM is not usable, PCR7 binding is not supported, Hardware Security Test Interface failed and device is not Modern Standby, Un-allowed DMA capable bus/device(s) detected, TPM is not usable
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	Yes
Hyper-V - Data Execution Protection	Yes

rtyr, 2021-03-10:
I was not able to reproduce it on W10 64bit.

mia-0, 2021-03-22:
I can reproduce this on openSUSE Tumbleweed, but with almost any file. 2.3.0 and git master crash right after G-Code generation for me. However, I do NOT get a crash if I use the `--loglevel` option with value 4 or greater. Awesome!

So I’ve made a few attempts at getting a backtrace (GDB needs a few GB of RAM for that with all the wx nonsense), but every time PrusaSlicer would hang instead of crash. However, when hitting Ctrl+C, it always seems to be stuck near `Slic3r::GLShaderProgram::set_uniform(char const*, std::array<float, 4ul> const&)`, so this is probably a grahics driver issue.

My GPU is an AMD Radeon RX Vega 54 with these software versions:
```GL version: 4.6 (Compatibility Profile) Mesa 20.3.4
Vendor: AMD
Renderer: Radeon RX Vega (VEGA10, DRM 3.40.0, 5.11.6-1-default, LLVM 11.0.1)
GLSL version: 4.60
```

Setting `LIBGL_ALWAYS_SOFTWARE=1` fixes the crashes (it just makes everything too slow to be usable). `LIBGL_DRI3_DISABLE=true` also does, without the performance penalty. Should probably report this to the driver devs…

I would not be surprised if whatever causes this is also tripping up Intel’s windows drivers, given how bad they are.

staffert, 2021-03-23:
...so I'm not crazy...

mia-0, 2021-03-23:
I’ve also noticed this warning:

```
dri_create_context: glthread isn't thread safe - missing call XInitThreads
```

That seems suspicious.

bubnikv, 2021-03-24:
Interesting.

pcsx2: call XInitThreads at init · ***@***.*** (github.com)
<https://github.com/PCSX2/pcsx2/commit/99180f5afbe012d87ef2cd68c3250e3dde90f396>

bubnikv, 2021-03-24:
@mia-0 When are you getting that warning? We are supposed to call OpenGL calls from the main thread only. Could you debug PrusaSlicer for us?

@enricoturri1966 do we call any OpenGL API from worker threads?

bubnikv, 2021-03-24:
We may need to call https://linux.die.net/man/3/xinitthreads
wxWidgets does not do that for us.

mia-0, 2021-03-24:
@bubnikv I get that warning twice at startup, before the main window is displayed.

I could help with debugging, sure! Going to try calling XInitThreads in an hour or two.

mia-0, 2021-03-24:
Alright, I did that. Warnings disappeard, crashes did not. What’s more, with this version (git master instead of 2.3.0), was also getting the crash with `LIBGL_DRI3_DISABLE=true`.

I also haven’t been able to reproduce it on my 2012 Haswell laptop (Intel HD4000-series) running Tumbleweed, or another desktop with Ryzen integrated graphics (Renoir). 

So… I remembered using adriconf at some point, and removed `.drirc` on the machine I was having trouble with. The problems disappeared. As expected, the culprit was `mesa_glthread` (allow offloading GL driver work to separate thread). I had that enabled in the default profile.

Apparently, this is not the bug we’re looking for. Sorry about the noise!

bubnikv, 2021-03-25:
@mia-0 Thanks for the report. Unfortunately the graphics driver situation on Linux is not stellar and we cannot support everything.

lukasmatena, 2021-04-15:
Can you guys please try out the new [2.3.1 patch release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1-rc) to see if it helps? It has in https://github.com/prusa3d/PrusaSlicer/issues/6345#issuecomment-820503201. Thanks.

bubnikv, 2021-10-05:
No response, closing.

closed 2021-10-05 by bubnikv