prusa3d/PrusaSlicer #6205: Slicer won't install

DubWil4126, 2021-03-10:
### Version
2.3.0

_Use `About->About PrusaSlicer` for release versions_
Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315 (GitHub version)
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 17,130MB

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Original I3 MK3

### Behavior
PrusaSlicer 2.3.0 won't install.  It had installed OK previously but after I uninstalled a previous version it was gone and I'm not able to reinstall it. steps:
[image]

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_ yes

The GitHub version runs but appears to install when launched and Fusion 360 doesn't work with it as it did with the normally installed version.  I get a popup message: "PLrusaSlicer 2.3.0 requires an active Internet connection for installation.  Please check your network configuration and proxy settings."  This doesn't help as I have no idea what it is expecting.
[image]
[image]

#### Project File (.3MF) where problem occurs: NA
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Janys, 2021-03-10:
Hello, thank you for reporting the issue. It seems like the installer might have trouble detecting an active internet connection. We're investigating the issue.

lukasmatena, 2024-02-18:
Closing as outdated. We are using a different installer now.

closed 2024-02-18 by lukasmatena