prusa3d/PrusaSlicer #6208: Feature Request: Custom Rafts/Brim

Animaljg, 2021-03-11:
### Version
2.3.0

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3s

### Behavior
I find that when printing a raft with a brim the print struggles to adhere due to the complex shape of the brim. 90 degree turns can be tricky for a 3d printer and often the brim is just hundreds of 90 degree turns in very short succession.

It would be nice if I could set a custom shape for the Raft or the Brim around supports to give straight edges or just a "Straighten Brim" option, or something to that effect.

_Is this a new feature request?_ YES

#### Project File (.3MF) where problem occurs
N/A

ishyca, 2021-03-12:
I am interested in this feature as well, a similar one to "Paint on support feature" if we have a "Paint on brim". 

reason why this is useful: 
example: printing a bulbausaur model 
1. save support material, I have overhang which are ok without support, but I want all 4 feet to be solidly attached. to the center support structure.  
2. save time have to do the same by using modifier, "paint on" feature is bit more intuitive. 
[image]

SphaeroX, 2021-05-06:
Paint on brim sounds fantastic, but if you take a closer look at wraping you can see where the problems are. 
Edges where the material cools down faster...

This occurs less on circular surfaces.

I have made an example here with the CAD programme, but it would be great if you could simply draw in brims.

[image]

SphaeroX, 2021-05-08:
saw this cool brim function on SuperSlic3r !

[image]

closed 2026-09-29 by prusaslicer-bot[bot]