prusa3d/PrusaSlicer #6209: Feature Request: Continous Counter for Output Filename

foreachthing, 2021-03-12:
### Behavior
* the new temp-file-method broke my post processing
* I need for pseudo version control an increasing number (prefix) in my output filename (004636_file.gcode). 

_Is this a new feature request?_
Yes, please. 

There's now way,  I can think of, to achieve it with my post processing... any ideas are still appreciated.

murk-sy, 2021-03-12:
Sequential numbering could still cause collisions if different instances of the slicer are used - which is unlikely, but wouldn't fix the problem universally.

This might not be optimal for your needs, but you can use `[timestamp]` in your output filename to get output like `20210312-144503` (date and time), so it is at least still sortable like a serial number would be. Unless you're slicing multiple files per second, collisions are very unlikely even with multiple instances of slicer being used at the same time.

foreachthing, 2021-03-12:
@murk-sy, this works well enough for the moment, I guess.
The issue I have with this is, that the timestamp is really long. When I choose my file on the display, it eats up almost the whole line (filename nearly not readable).

> Sequential numbering could still cause collisions if different instances of the slicer are used - which is unlikely, but wouldn't fix the problem universally.

Possibly you're right. That's why the post processing script worked so good. It could've handled that, most likely ;-)

foreachthing, 2021-10-06:
Solved with post processing (https://github.com/foreachthing/Slic3rPostProcessing/blob/master/SPP-Python/Slic3rPostProcessor.py).
Also added reverse counter (count down: https://github.com/prusa3d/PrusaSlicer/issues/6042#issuecomment-915390686)