prusa3d/PrusaSlicer #6210: Add a "0" option to "Speed for non-print moves" to allow use of values from "Machine limits"

Mjankor, 2021-03-12:
### Version
2.3.0+

### Operating system type + version
OS X - Big Sur

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S & BNBSX extruder
Prusa Mk3S+ & BNBSX extruder
Custom CoreXY

### Behavior
NA (Feature request)

_Is this a new feature request?_
Yes

I'd like to request that "Speed for non-print moves", if set to zero, should use the machine values specified in the machine limits (Maximum feedrate X and Y). This would allow the use of different travel velocities on different machines, without affecting the rest of the (typically filament dependant) print speed settings.

rtyr, 2021-04-14:
Machine limits are available only for Marlin g-code flavor.

rtyr, 2021-06-17:
No response. Machine limits are now available only for one g-code flavor, so it would not make much sense as global setting. Closing.

closed 2021-06-17 by rtyr