prusa3d/PrusaSlicer #6213: Extruder Clearance Improvement

colatuttorf, 2021-03-12:
### Version
2.3.0+

### Operating system type + version
MacOS 10.13.6

### 3D printer brand / version + firmware version (if known)
Issue exists for all printers in PrusaSlicer 2.3.0+

### Behavior
* _Describe the problem_
Extruder Clearance is defined as a regular cylinder
* _Steps needed to reproduce the problem_
Enable Complete individual objects and specify extruder clearance as a radius
* _Expected Results_
Expect to be able to include actual detail in the extruder definition, ie: a bounding box in x,y coordinates and where in that box/offset the extruder is
* _Actual Results_
only a radius is able to be entered and the slicer assumes that the extruder is dead center located in a cylinder, resulting in substantially less useful sequential printing mode, particularly on smaller printers

_Is this a new feature request?_
Yes

Instead of Radius, would like to be able to input as such:
[image]

Photogad, 2021-03-15:
I also would like to see this.  I print a lot of sequential objects and this wouod be useful. I have a nozzleCam  mounted behind my extruder so the default cylinder doesn't really work well.

ledvinap, 2021-03-15:
This may be implemented as clearance part modifier - specify volume that must be empty when printing next part in sequence. Then simply test if there is collision with already printed parts.
Clearance volume can be quite easily prepared as Minkowski sum of object and extruder (probably in external program).

closed 2026-09-29 by prusaslicer-bot[bot]