prusa3d/PrusaSlicer #6217: Object prints over the skirt

jkuusama, 2021-03-13:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Prusa MK2S (irrelevant, slicer issue)

### Behavior
* Sometimes on crowded plate, the skirt and parts overlap
* Slice the attached project
* skirt away from the objects; if not enough room, an error message
* skirt under object

[image]

#### Project File (.3MF) where problem occurs

[up looking top_7.zip](https://github.com/prusa3d/PrusaSlicer/files/6134946/up.looking.top_7.zip)

jmd73, 2021-03-13:
@jkuusama Not all objects are flat on the bed. When
When I open your AMF PrusaSlicer even warns about it.

[image]

And you can see it when I have supports enabled.

[image]

If you flip them so the other other side is down, you can see why they are not flat on the bed:

[image]

And after they are flipped, everything should be fine.

[image]

Best regards, Another Juha. :)

rtyr, 2021-03-13:
Some parts are upside-down. Closing.

[image]

closed 2021-03-13 by rtyr