prusa3d/PrusaSlicer #6218: Button tooltip in the way

mihai-stanimir, 2021-03-13:
### Version
_2.3.0_

### Operating system type + version
_Windows 10_

### 3D printer brand / version + firmware version (if known)

### Behavior
When clicking "send to printer" the button tooltip shows up and I click on it instead of clicking on the button. Happens about a third of the time.
[image]
Can this tooltip be disabled?

jojowasher, 2021-03-21:
This also happens when the Slice Now is visible, it is extremely annoying because I think I have clicked slice and wait for it to be done and nothing happens.

pixsrv, 2021-07-22:
it does't have to be disabled
it can be repositioned to not cover the meaningful item
or
it can just not consume this click
so the button can receive this event

bubnikv, 2021-10-04:
This is unfortunately a general issue of the tooltips on Windows. I posted a request on wxWidgets forum, wxWidgets is the UI framework that wraps around Win32 UI framework.

bubnikv, 2021-12-08:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/e2f2ed41720c73170c0e08f8650394d253cae973 in PrusaSlicer 2.4.0-beta3.

closed 2021-12-08 by bubnikv

tonycstech, 2024-05-27:
I am still having issue with it.
Tooltip pops out in GCODE section for the machine setup and NEVER goes away covering half the screen,
Need to be able to DISABLE IT 100%

5jvm0u4, 2025-01-06:
@bubnikv Still having this issue in 2.9.0, please re-open this issue.
[image]
Windows 24H2

ungutknut, 2025-03-07:
Just trying to switch from Cura to PrusaSlicer and the extensive/obstructive tooltip texts are a major turnoff for me. Not only do they cover all kind of buttons and text boxes, but sometimes they just won't disappear (e.g. at the start-gcode text box) or flicker like crazy when moving the mouse (e.g. just hover horizontally over the leftside "layer height" text at the layer height setting).

Possible solutions:

1) implement an option to disable them completely (which I wouldn't prefer since they're really helpful)
2) dedicate some separate screenspace for these texts (see cura for reference). The bottom left section of the settings "tree" is left unused, that'd be a good candidate
3) increase delay for tooltip text to appear. There seems to be the windows-typical delay most of the time, but sometimes it appears instantly (mostly when hovering the [flickering] leftside settings names) which is especially annoying.

Also please re-open this issue.

5jvm0u4, 2025-03-07:
@ungutknut @tonycstech you could leave some comment here #14243