prusa3d/PrusaSlicer #6219: Incorrect slice with STL-File

Bilbo707, 2021-03-13:
### Version
2.3.0+win64

### Operating system type + version
Windows 10 - 20H2

### 3D printer brand / version + firmware version (if known)
Prus i3 MS3S+

### Behavior
I get a replacement for my Braun Multiquick Minipimer (https://www.thingiverse.com/thing:3088323).
When I pass the slicer it fills the lower part completely, so the printed part fits not with the Braun Minipimer.
With Cura 4.8.0 the stl-file will be sliced correct.

#### Project File (.3MF) where problem occurs
[files.zip](https://github.com/prusa3d/PrusaSlicer/files/6135209/files.zip)

rtyr, 2021-03-13:
The model is broken. It is not completely OK in Cura either.

Bilbo707, 2021-03-14:
Thank you for the quick response.
How can i check if a model file is broken?  Is there a tool to check STL-Files?
In my case the Slicer did not show a warning.

regards

Paps2020, 2021-03-15:
[files2.zip](https://github.com/prusa3d/PrusaSlicer/files/6139218/files2.zip)

Bilbo707, 2021-03-15:
Thank you. for your comment.

But from the top-side view it looks like my Slicer work. 
If you turn the sliced model upside down you will see the error.  The lower part is filled completely, so the printed part fits never with the Braun Minipimer.

closed 2026-09-29 by prusaslicer-bot[bot]