prusa3d/PrusaSlicer #6223: paint-on support - 2 problems

3Dietrich, 2021-03-14:
### Version
Slicer 2.3.0
### Operating system type + version
MacOS 10.15.7
### 3D printer brand / version + firmware version (if known)
i3mk3s
### Behavior
#### _Describe the problem_
I think this behavior is wrong although I can't imagine that this went through your QA so unseen.
2 problems:
1. On the left side is well visible blue area. There is nothing of it in the PrintPreview (I know this makes not really sense, but who knows, and the output "nothing" is not a smart decision either)
2. on the right side the blue is drawn deeper from the edge to shorten the bridge. In the PrintPreview it is ignored and is only on the outside.
3. well.. the 3rd problem I just found out: It can NOT build any bigger support with *any Slicer function*!! I tried "Everywhere", adding a "support enforcer", and of course "paint-on" - paintig all blue - nothing does make the support bigger! There are Filaments with less bridging quality - Slicer doesn't support this right now...

what I want in "paint-on support":
[image]
what I got:
[image]
#### _Steps needed to reproduce the problem_
I don't know what I did extraordinary. I clicked "Paint-on Support", made the 3d mouse something bigger and painted what you can see in the screenshots.
3mf attached
#### _Expected Results_
I expected on both sides the support at the blue painted areas
#### _Actual Results_
well.. you can see: left is no support although where it should be, right is on the outer corners but I painted far deeper...
screenshots in "Describe the problem"
#### Project File (.3MF) where problem occurs
Here ist the m3f of the thing shown in screenshot:
[BlumenTopf.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6137082/BlumenTopf.3mf.zip)

neophyl, 2021-03-14:
I haven’t had chance to download your zip as I’m im on my iPad at the moment but I would hazard a guess that you have “don’t support bridges option” ticked. It’s putting support where you painted it on the perimeters and all the interiors section is bridged, as you can see from the preview colour (bridge infill colour). As you have painted any support enforcer on the other perimeter section no support is there obviously. 
This is normal for Prusa Slicer.

3Dietrich, 2021-03-14:
yes!
Thank you.
wow, what a trap

closed 2021-03-14 by 3Dietrich