prusa3d/PrusaSlicer #6225: Feature Request: Filament Overrides depend on Active Printer Profile

bepstein111, 2021-03-14:
### Version
2.3.0+win64

### Operating system type + version
Windows 10 x64 (but really all of them)

### 3D printer brand / version + firmware version (if known)
Creality Ender-3 and CR-10, both running Marlin 2.0.7.2

### Behavior
The addition of filament overrides for retraction settings is a huge improvement, and solved the problem of "retraction settings also depend on material being used, not just the extrusion chain in place," however for users of PrusaSlicer with multiple printer profiles, there's no easy way (afaik) to automatically change these overrides when you switch printer profiles. In my case I've got one printer set-up for direct drive, the other is bowden, so retraction settings can vary A LOT between the two printers. The only current alternative is to set up duplicate profiles for each printer with correct dependencies and different retraction settings, which just feels like work the software should be able to handle.

Essentially what I'm asking for is for the current contents of the Filament Overrides menu (in Filament Settings panel) to be duplicated for each physical printer configured in some way. That's how I imagine solving this problem at least, I can imagine multiple ways for that to happen, and I am very open to other solutions as well!

_Is this a new feature request?_
Yes

### Project File (.3MF) where problem occurs
All in which multiple printers are configured

bepstein111, 2021-03-14:
forgive me if this is duplicated somewhere, I looked but couldn't find anything that looked like exactly this.

Photogad, 2021-03-15:
I do like this idea. Also would like to see fan speed and cooling overrides in the print profiles (not just in filament profiles) so I can disable auto cooling in my "speed" profiles without having to make multiple filament profiles

rtyr, 2021-03-15:
> The only current alternative is to set up duplicate profiles for each printer with correct dependencies and different retraction settings.

This is actually correct solution and it basically does what do you request. I can understand, that you need only retraction settings in your specific case, but in general, the filament settings can be completely different for different printer (temperatures, linear advance, custom g-code, etc.). Configuring proposed "links/overrides" would be probably even more complicated (from user perspective) than creating new filament profile visible only for specific printer.

closed 2026-09-29 by prusaslicer-bot[bot]