prusa3d/PrusaSlicer #6226: Acceleration Control setting does not seem to work?

Photogad, 2021-03-15:
### Version
2.3.0

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
MK3S, latest firmware 3.9.3

### Behavior

Acceleration control in the print profiles does not seem to work. Acceleration always seems to stay for the "maximum" setting in the physical printer profile. See case examples below.

**Prusa default acceleration control**
[image]

**Prusa default maximum acceleration**
[image]

**My maximum acceleration**
[image]

### Example

[image]

Cube on the left is with prusa default profiles (and perimeter acceleration control set to default of 800)

Cube on the right is with maximum acceleration set to 8000  (and perimeter acceleration control set to default of 800)

You can see the ringing /ghosting on the cube on the right. But there should be no ringing and ghosting on the right cube because it should be using 800 acceleration for both cubes, not 800 and 8000.

### Expected Behavior

I would like it to use the acceleration control settings I set, so I can fine-tune acceleration and not just use the maximum acceleration control for everything. I would expect maximum acceleration to be a hard-limit on all values, not the actual value it uses. I want perimeters to stay with low acceleration but rest of print to use high acceleration. But I cannot seem to do this using the settings because it uses maximum acceleration control for everything.

rtyr, 2021-03-15:
Please provide both g-codes and 3mf project file.

Photogad, 2021-03-15:
> 
> 
> Please provide both g-codes and 3mf project file.

No problem. Find the file attached. There's one gcode where everything is at the prusa defaults for acceleration. And another where I changed the maximum acceleration limits to 8000 but kept the acceleration control to the defaults.

You'd expect it would use the acceleration control values for the prints, not the maximum values. But it uses the maximum values in a real print.

I looked at the gcode --- I see in there where it's putting the maximum values in the gocde, but I see no where in the gcode that it is putting the "acceleration control" values... unless I'm missing it.

[Accels.zip](https://github.com/prusa3d/PrusaSlicer/files/6143098/Accels.zip)

rtyr, 2021-03-15:
I don't see any issue in your g-codes. It reflects your settings.

"Max Accel Only_0.2mm_PLA_MK3SMMU2S_11m.gcode" - `M204 S` commands based on acceleration control are there. No problem.

"Prusa Default_0.2mm_PLA_MK3SMMU2S_10m.gcode" - It was not sliced with the default print profile (provided 3MF was not used for this gcode). All acceleration values in acceleration control were changed to 5000, while max limit in the machine limits was left at 1250. The result is single `M204 S1250` command, that is correct.

Photogad, 2021-03-15:
@rtyr 
> "Prusa Default_0.2mm_PLA_MK3SMMU2S_10m.gcode" - It was not sliced with the default print profile (provided 3MF was not used for this gcode). All acceleration values in acceleration control were [...]

I'm not sure what happened why the gcode didn't match the 3MF. I will re-export the gcode and check again

but the acceleration were certainly not changed to 5000.... they are at the defaults (see below)

[image]

Photogad, 2021-03-15:
@rtyr 
Exported G-code again.

[Desktop.zip](https://github.com/prusa3d/PrusaSlicer/files/6143510/Desktop.zip)

**Default2.gcode and .3mf should be using default settings as per this:**

[image]
[image]

**MaxAccel2 gcode and 3mf should be using only 8000 maximum acceleration on the printer tab as per follows, but default acceleration control as shown:**

[image]
[image]

rtyr, 2021-03-15:
> but the acceleration were certainly not changed to 5000....

Check the comments at the end of the provided g-code (example `; infill_acceleration = 5000`) or import import the settings to the PrusaSlicer by using `File - Import - Import config...` (it is possible to import the settings from the g-code).

rtyr, 2021-03-15:
`Exported G-code again.`

Both g-codes are OK. Search for `M204 S` commands.

rtyr, 2021-03-15:
My guess is, that the right cube on the photo was printed with that first "prusa_default" g-code (1250 acc instead of default 800).

Photogad, 2021-03-15:
> 
> 
> My guess is, that the right cube on the photo was printed with that first "prusa_default" g-code (1250 acc instead of default 800).

I'm going to reprint them with the new gcodes now to be sure.

Photogad, 2021-03-15:
@rtyr 

Ok, still same results. Printed with above gcodes, and the maximum acceleration @ 8000 file still has horrible ghosting and ringing even though the "acceleration control" remains the same as the prusa default of 800/1000/1000/1000/1000

So I think, either PrusaSlicer is not outputting the correct gcode M204 to a way the printer can understand it (even though we see it in the gcode script), or it's a firmware issue where the printer is ignoring the subsequent M204 gcode after the first M204 lines where it establishes the maximum acceleration rates.

Photogad, 2021-03-15:
Or maybe I just completely am not understanding how acceleration control works.

Is acceleration control what the printer will absolutely use ??? Or does the printer use acceleration control as a base (minimum) setting and scale it up to the maximum setting?

rtyr, 2021-03-15:
This is the command generated before start g-code. It is based on the machine limits:

`M204 P8000 R1250 T8000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2`

This is the command generated based on the acceleration control (placed before each feature, for example perimeters, infill, etc.):

`M204 S800 (value is based on the feature).`

`M204 S` overrides initial `M204 P T` command.  Initial M204 P8000 xxx xxx is no longer valid before initial skirt is printed.

Anyway, I printed both gcodes on MK3S and I don't see any difference. Output from the PS is OK, firmware behavior seems also OK and print results are same as expected.

Btw. there is a hardcoded max acc limit in the firmware (2500 for normal mode, 960 for stealthchop).

Photogad, 2021-03-17:
> Anyway, I printed both gcodes on MK3S and I don't see any difference. Output from the PS is OK, firmware behavior seems also OK and print results are same as expected.
> 

Ok very bizarre,.. I wonder if anyone else can test. With the 2nd gcode file I definitely do get a lot more noticable ringing/ghostin (well you saw my photos).

You are on 3.9.3 prusa firmware as well, or another printer?

rtyr, 2021-03-17:
2 different MK3S printers with 3.9.3, build 3556, directly from SD card. I also tried M204/M503 FW behavior through pronterface and found nothing suspicious.

Both g-codes use same acceleration values, so there shouldn't be any reason for different results. Only initial M204 command is different, but it is overriden at line 80. If you get different results on your printer, it is not PrusaSlicer/g-code related issue.

> well you saw my photos

That photo is not related to those latest g-codes if I am not mistaken. The g-codes you provided initially actually used different accelerations. Also you have to look from the exactly same angle and with the same light.

rtyr, 2021-03-22:
I am closing this issue, the g-code generated by PS is OK. Feel free to open an issue at Prusa-Firmware, but as I mentioned I wasn't able to reproduce it.

closed 2021-03-22 by rtyr

terracedr, 2021-05-30:
I am just wondering if any further results on the thread actually have been discovered? I have been wondering abiout all this as well as I too hav a slight ghosting problem as the OP. All with the similar settings but not using 8000 max acc. just the default of 1000 Max and 800 in the print settings.
[image]

Henk72, 2022-12-02:
To me it seems the G-code is correct with acceleration control. Unfortunately the slicer does not limit the speed as a result of lowered acceleration. If you reduce acceleration to 10mm/s2 (crazy low), the speed of the print seems to be the samen. Of course the actual print will be slower.

Henk72, 2022-12-02:
So the simulation results on screen in the slicer are not correct, the g-code however is. It adds the M204 command. Too bad the slicer does not calculate the lowered speed due to the limited acceleration.

rtyr, 2022-12-02:
Time estimator simulates the firmware, so accelerations are taken into account. It is not visualized in the preview obviously.