prusa3d/PrusaSlicer #6229: Feature Request: Infill tilt as an advanced option

martincizek, 2021-03-15:
### Version
2.3.0

### Use Case
Can share more details if it is relevant. I came across a project where:
* the model had to be moderately rotated around the X axis for printing
* the print was required to withstand high compressive stress in the former (unrotated) Z axis.

The option of achieving infill oriented along the tilted Z axis by tilting the infill itself proved to be working even in the real world and it was quite easy to achieve by patching PrusaSlicer, see below.

### Feature Request
Allow tilting the infill using two (very advanced) angle parameters - something like "yaw" and "pitch".

My question is - can this be of any general use? I can try to generalize it and submit a PR, but placing this to GUI meaningfully would be a challenge, as I indeed understand it would be rarely used.

In fact, my biggest issue is that upgrading PrusaSlicer would become a pain, so any alternative suggestion e.g. on modularizing infill patterns would be nice.

### Proof of Concept
It can be easily achieved by changing the *reference point* over `z` position in overriden `_infill_direction()` method. Sharing the code below. The parameters are hardcoded - the "yaw" angle is 90 degrees and "pitch" is 13 degrees.

Successfully tested with patterns from `FillGrid` and `FillTriangles`, but I guess that any 2D infill pattern should work. I've also verified that the infill pattern is generated correctly even if the reference point drifts outside of the object.

``` c++
class FillTilted : public FillRectilinear
{
public:
    Fill* clone() const override { return new FillTilted(*this); }
    ~FillTilted() override = default;
    Polylines fill_surface(const Surface *surface, const FillParams &params) override;

    static constexpr float CorrectionAngle = 0.;
    static constexpr float InclineAngle = 13.;

protected:
    // The grid fill will keep the angle constant between the layers, see the implementation of Slic3r::Fill.
    float _layer_angle(size_t idx) const override { return CorrectionAngle; }
    std::pair<float, Point> _infill_direction(const Surface *surface) const override
    {
        std::pair<float, Point> dir = FillRectilinear::_infill_direction(surface);
        dir.second.y() += scale_(z * tan(Geometry::deg2rad(InclineAngle)));
        return dir;
    };
};
```

### PoC Results

Demo prism with tilted triangular infill (not the real use case):
[image]

Demo prism with tilted triangular infill and perimeters hidden:
[image]

Demo prism with normal triangular infill and perimeters hidden:
[image]

lukasmatena, 2021-03-16:
@martincizek I will not comment on the necessity of such a feature, just FYI: your suggestion is not a tilt, it is a shear. It will certainly fail badly for angles not close to vertical, probably even for the simplest infill types. Doing it properly would be much much more complicated.

martincizek, 2021-03-16:
> just FYI: your suggestion is not a tilt, it is a shear.

Thanks for this! I've spent quite some time looking for the correct name and having the right keyword is invaluable. :)

> It will certainly fail badly for angles not close to vertical, probably even for the simplest infill types.

Totally understand. With my 13 degrees, it seems to give the mechanical benefit I expected. If it were 30 degrees, it would be likely counterproductive and using a 3D infill pattern would be better.

> I will not comment on the necessity of such a feature, [...] Doing it properly would be much much more complicated.

Indeed. :) Still, I'd like to share this somehow. Is the Prusa forum the right place?  Or is there some "datababase of excentric PrusaSlicer hacks"? :)

P.S. Does it make sense to create a ticket for infill plugin support for those who want to code their own fills? I've only found [this forum entry](https://forum.prusaprinters.org/forum/prusaslicer/create-your-own-infill-patterns/), but no relevant related issue.

closed 2026-09-29 by prusaslicer-bot[bot]