prusa3d/PrusaSlicer #6230: Bed not turning off by default for custom printer, Missing M140 in end gcode

Jmlannan, 2021-03-16:
### Version

Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Custom Marlin Printer

### Behavior
I am not entirely sure how to replicate but when setting up my custom printer I ended up not having a M140 S0 in the end gcode to turn of the bed heater leaving it on when the print finished. This is pretty dangerous behavior and I think there should probably be some check somewhere to make sure the heated bed isn't left on.

#### Project File (.3MF) where problem occurs
Config files:
[PrusaSlicer_config_bundle.ini.txt](https://github.com/prusa3d/PrusaSlicer/files/6145187/PrusaSlicer_config_bundle.ini.txt)

lukasmatena, 2021-03-16:
Duplicate of #2698 and #644. Closing this one.
I myself think that everyone can add the turn-off command into their custom end gcode without forcing the same behavior to the others. Also if it is "pretty dangerous" to turn the bed heating on, it should probably stay off even during the print. But this should really be discussed in #644.

closed 2021-03-16 by lukasmatena

Jmlannan, 2021-03-16:
@lukasmatena  Perhaps not dangerous but unexpected behavior like that means I had no indication that the hot bed was still on after a print finished and I had no reason to expect that it would be. When I configure a new prusa printer the heat bed is turned off automatically and I would expect a custom printer to do the same. It seems strange to me that a user would be expected to turn this off by having to edit the end gcode.