prusa3d/PrusaSlicer #6234: [BUG] No top solid infill when apply ironing modifier

amper128, 2021-03-16:
### Version
Version 2.3.0+linux-x64-GTK3

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
Any printer

### Behavior
I want 10 top solid layers, but when add "disable ironing" modifier - i got zero top solid layers.

Steps to reproduce:
1. add 20x20x20 box
2. rotate around X to 335deg
[image]
3. cut Z to 10mm and keep upper part
4. cut Z to 12mm and keep lower part
[image]
5. enable ironing on print settings
[image]
6. set horizontal shells: top/bottom solid layers to 10 (or any big value)
[image]
7. add modifier box: 20x20x20, xy offset to 0, z offset 4
[image]
8. add settings "ironing" -> "enable ironing"
[image]
9. uncheck "enable ironing"
[image]
10. slice
11. scroll layers down
[image]

No solid infill before ironing layer!

rtyr, 2021-03-17:
Please provide your 3MF project file.

amper128, 2021-03-17:
[ironing modifier bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6158398/ironing.modifier.bug.3mf.zip)
remove .zip from filename

rtyr, 2021-03-18:
I think it is expected behavior with current implementation. If the infill setting isn't the same, it will separate the areas and slice them separately. This is side effect of it.

It should be fine if you place the modifier only where you actually want to disable ironing or in this case, it would be better to just use Height range modifier from 8-12.38.

amper128, 2021-03-18:
Ookay, this is feature, not bug...

bubnikv, 2022-01-03:
Isn't it the same issue as https://github.com/prusa3d/PrusaSlicer/issues/7077#issuecomment-991849141 ?

closed 2026-09-29 by prusaslicer-bot[bot]