prusa3d/PrusaSlicer #6238: Crash while preview gcode when tool position is 0

PuddingBaer91, 2021-03-18:
I am using windows 10. If you slice a part and enable "show travels" in the gcode preview the tool will crash without an error if you move the tool position slider to 0

rtyr, 2021-03-18:
Please provide your 3MF project file.

PuddingBaer91, 2021-03-18:
> Please provide your 3MF project file.

It's not needed because it happens with all .stl flies i slice. (Just slice anything with stock mk3 settings)

PuddingBaer91, 2021-03-18:
> I am not able to reproduce it.

Hmm strange. It's also present after a complete reinstall of Windows. Also present with superslicer. Does a video of the crash will help?

PuddingBaer91, 2021-03-18:
[bow_slider.zip.zip](https://github.com/prusa3d/PrusaSlicer/files/6167293/bow_slider.zip.zip)

please have a look:
https://photos.app.goo.gl/9BMRqmFAEmFZ3ZbW8

fe60, 2021-03-19:
I never faced such a crash and it's not reproducible for me. Not even with the shared project file.
To get more information run the slicer from command line activating the maximum log level and share the log.

1. Go to the installation folder of the slicer in the Windows Explorer
2. Move the mouse pointer on the white area in the Explorer
3. Press shift and the the right mouse button
4. Select `Open PowerShell here` in context menu (entry only appears when Shift is pressed)
5. Paste the command below into Powershell (copy and then right click in Powershell)
6. Get the Application crashing
7. Attach the log file created on your desktop

`.\prusa-slicer-console.exe --loglevel=5 *>$env:USERPROFILE\Desktop\PrusaSlicerLog.txt`
Runs slicer with maximum logging and redirects all outputs into a file named PrusaSlicerLog.txt on your Desktop.

PuddingBaer91, 2021-03-19:
`Name der fehlerhaften Anwendung: prusa-slicer-console.exe, Version: 2.3.0.0, Zeitstempel: 0x5ffc50a6
Name des fehlerhaften Moduls: nvoglv64.dll, Version: 23.21.13.9125, Zeitstempel: 0x5aab74c2
Ausnahmecode: 0xc0000005
Fehleroffset: 0x0000000000e939c9
ID des fehlerhaften Prozesses: 0x3d04
Startzeit der fehlerhaften Anwendung: 0x01d71c9851ab9f07
Pfad der fehlerhaften Anwendung: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer-console.exe
Pfad des fehlerhaften Moduls: C:\Windows\System32\DriverStore\FileRepository\nvdw.inf_amd64_455f3ef30558a66d\nvoglv64.dll
Berichtskennung: b3cb4f4b-fe69-48bc-8b17-15466dd918d6
Vollständiger Name des fehlerhaften Pakets: 
Anwendungs-ID, die relativ zum fehlerhaften Paket ist: 
[PrusaSlicerLog.txt](https://github.com/prusa3d/PrusaSlicer/files/6169600/PrusaSlicerLog.txt)
`

PuddingBaer91, 2021-03-19:
[PrusaSlicerLog.txt](https://github.com/prusa3d/PrusaSlicer/files/6169601/PrusaSlicerLog.txt)

this is what the windows events showing:
`  Name der fehlerhaften Anwendung: prusa-slicer-console.exe, Version: 2.3.0.0, Zeitstempel: 0x5ffc50a6
Name des fehlerhaften Moduls: nvoglv64.dll, Version: 23.21.13.9125, Zeitstempel: 0x5aab74c2
Ausnahmecode: 0xc0000005
Fehleroffset: 0x0000000000e939c9
ID des fehlerhaften Prozesses: 0x3d04
Startzeit der fehlerhaften Anwendung: 0x01d71c9851ab9f07
Pfad der fehlerhaften Anwendung: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer-console.exe
Pfad des fehlerhaften Moduls: C:\Windows\System32\DriverStore\FileRepository\nvdw.inf_amd64_455f3ef30558a66d\nvoglv64.dll
Berichtskennung: b3cb4f4b-fe69-48bc-8b17-15466dd918d6
Vollständiger Name des fehlerhaften Pakets: 
Anwendungs-ID, die relativ zum fehlerhaften Paket ist:      `

fe60, 2021-03-24:
It seems the most important part didn't reach the log file. Nevertheless you added it.
I'm not an expert for such an analysis, but I asked the Internet for help. I found a thread in a support forum recommending to update the graphics driver. The affected user faced issues with the driver version running your machine and an update solved the problem.
Did you try to update the NVIDIA graphics driver?

https://us.forums.blizzard.com/en/sc2/t/majere-crashing-on-launch/12788/7

PuddingBaer91, 2021-03-24:
> It seems the most important part didn't reach the log file. Nevertheless you added it.
> I'm not an expert for such an analysis, but I asked the Internet for help. I found a thread in a support forum [...]

Thank you. I already tried some different drivers but no luck. May because i use a workstation graphic card (QUADRO P2000) it uses openGl. I just wanted to tell you the bug but when it is not your "fault" you can't fix it. For me it works, i just have to watch out not to go to zero with the tool position.. so thank you for the help. I think you can close this. If i find a solution i will post it here. Thank you

bubnikv, 2021-10-05:
great, closing

closed 2021-10-05 by bubnikv

bubnikv, 2021-12-14:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/7429#issuecomment-992951495