prusa3d/PrusaSlicer #6239: STL file doesn't slice correctly - a part is missing

tomas-mazak, 2021-03-18:
### Version
```
Version 2.3.0
```

### Operating system type + version
```
tomas@arkham:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal
tomas@arkham:~$ uname -a
Linux arkham 5.4.0-66-generic #74-Ubuntu SMP Wed Jan 27 22:54:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

### 3D printer brand / version + firmware version (if known)
```
Prusa i2 MK3S+
```
### Behavior

I am trying to print the attached STL (sink strainer with a handle. After importing STL, model looks perfectly fine (3D editor view). After slicing, however, in the preview, the central handle is completely missing.

#### Project File (.3MF) where problem occurs

[bugreport.zip](https://github.com/prusa3d/PrusaSlicer/files/6166419/bugreport.zip)

neophyl, 2021-03-18:
Once again a openscad model that doesnt print correctly.  What a surprise.  The model has multiple errors.  104 Intersecting Faces, 1968 Zero Faces, 896 Zero Edges. but at least its manifold which is better than usual.  I generated the report using Blenders 3dPrint tool.  
I also used the 1 click fix button in the tool.  Afterwards it still has some issues but now it slices correctly.  
[image]
original on the left, fixed on the right

I've zipped up the fixed version for you.

It is very easy to generate models that aren't correct with Openscad.  

[sink_strainer_neo.zip](https://github.com/prusa3d/PrusaSlicer/files/6167075/sink_strainer_neo.zip)

rtyr, 2021-03-19:
Already fixed with https://github.com/prusa3d/PrusaSlicer/commit/9c5dd358ab23e9351dce0b9c71eaa5ddf92d6f17. Closing.

```
Fixing model import of models with nearly degenerate triangles, which
degenerate after transformation (translation or rotation).

There was certainly a bug in PrusaSlicer, however this bug was triggered
mostly by the ASCII STL files generated by OpenSCAD, likely due to insufficient
number of decimal digits when saving an ASCII STL, STL export should contain single precision floats
```

closed 2021-03-19 by rtyr