prusa3d/PrusaSlicer #6240: support blocker enforcer issue

fbujold, 2021-03-19:
### Version
 2.3.0+64
### Operating system type + version
windows 10
### 3D printer brand / version + firmware version (if known)
MK3S, MK2.5S

### Behavior
I want to support holes on a vertical wall ONLY
I done not want to have the support all the way to the bed surface
tried painter blocker enforcer and combination of support always goes to platform.
[Downloads.zip](https://github.com/prusa3d/PrusaSlicer/files/6174102/Downloads.zip)

_Is this a new feature request?_
Not a feature but a bug

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-03-22:
[image]
[image]

bubnikv, 2021-03-22:
Testing snug supports 

[image]

bubnikv, 2021-03-23:
Fixed with af9c7c967f08fcf22ffa628c324c90eda65230f1

    Implementing a new switch for the shape of support towers:
    expanded to a grid (the old way) vs.
    snug (like the upstream Slic3r, Cura or Ideamaker).
    
    Snug supports suffered from the degeneracies when merging overhang islands
    over a large number of layers when projecting the support towers down.
    We borrowed the idea & a bit of code from Cura by simplifying the support
    polygons by closing the concave cracks, see the smooth_outward() function
    and the MutablePolygon class.

closed 2021-03-23 by bubnikv