prusa3d/PrusaSlicer #6247: A model doesn't get sliced properly but the 3D editor view is good

ctheroux, 2021-03-21:
### Version
Version 2.3.0+linux-x64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 20.04.2 LTS
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.3.0+linux-x64
Build:     PrusaSlicer-2.3.0+linux-x64-202101111322

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.4.0-67-generic x86_64
Total RAM size [MB]: 16,734MB
OpenGL installation
GL version:   4.6.0 NVIDIA 460.39
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 760/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
[... 309 lines cut ...]
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
ORIGINAL PRUSA I3 MK3S 3.9.0-3421

### Behavior
* _Describe the problem_
* When the attached file is loaded in PrusaSlicer, the 3D editor view is fine. But when the model is sliced, the hole under the half of sphere is blocked when looked in the Preview mode in PrusaSlicer.  I tried to print it and it is wrong also.
* _Steps needed to reproduce the problem_
* Load the model in PrusaSlicer.  Look at the hole at the bottom of the model in 3D Editor view mode.
* Slice the model.  Now look at the hole at the bottom in Preview mode or look at it at different layer.
[cover-small.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6177819/cover-small.3mf.zip)
[cover-small.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/6177821/cover-small.stl.zip)

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  * The print is wrong also.
[Previews.zip](https://github.com/prusa3d/PrusaSlicer/files/6177827/Previews.zip)

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

ctheroux, 2021-03-21:
Hi, additional information:  Works fine with C**a.

neophyl, 2021-03-21:
Just because it looks ok on the plater does not mean it actually is.  That model has 73 non manifold edges according to Blenders 3dprint tool report.  Therefore the model is broken.  
When I loaded your project into PS2.3 on Windows it said there were 96 errors that it fixed (48 degenerate facets and 48 facets removed .  I then sliced it and it actually sliced fine.  With the hollow hemisphere section.  BTW all the non manifold issues are around the bottom of the hemisphere.   Have you tried loading in your own project and slicing it ?

If I load the original model in then I get the same broken slicing you do.  Also PS does NOT report any errors with the original STL file.  Thats the real issue here, that PS is not reporting any problems with a file that definitely does have issues.  It looks like the save to 3mf has allowed PS to detect errors and fix them when reloaded.

The other option is to fix the underlying model.  Either through something like netfabb or a similar method of your choosing.

closed 2021-03-21 by ctheroux