prusa3d/PrusaSlicer #6253: Cancel slicing takes too long

zakutin, 2021-03-22:
### Version
2.3.0+win64

### Operating system type + version
Microsoft Windows 10 Pro (Version 10.0.19042 Build 19042)

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* _Describe the problem_
Cancel the slicing takes too long, when slicing with 10um resolution.
* _Steps needed to reproduce the problem_
During the slicing click Cancel.
* _Expected Results_
Cancel the slicing shall happen instantly, completely dismissing all background processes without any delay. Cancel is cancel. No wait.
* _Actual Results_
Cancel process lasts 1-2 minutes, depending on part size.

_Is this a new feature request?_
No.

bubnikv, 2021-03-22:
We need your model and ideally we need to know the slicing step at which
you believe the cancellation takes too long.

zakutin, 2021-03-24:
I'm using SLA inherited settings, 0.01 mm slicing resolution and supports.

I recorded two videos for info:
- [Version 2.1.1](https://www.youtube.com/watch?v=de52TJD02PU)
- [Version 2.3.0](https://www.youtube.com/watch?v=lnfWQETk33c)

Model files, STL and 3mf:
[model.zip](https://github.com/prusa3d/PrusaSlicer/files/6197185/model.zip)

bubnikv, 2021-03-24:
@tamasmeszaros would you please look at it? It looks as if the "Rasterizing layers" step could not be canceled in PrusaSlicer 2.3.0.

closed 2021-04-08 by tamasmeszaros

zakutin, 2021-06-02:
It looks like the issue still persists in Version 2.3.1+win64. It indeed happens on "rasterizing layers" step. Please see the videos below.
Big part

https://user-images.githubusercontent.com/53969222/120490443-b26fc400-c3b8-11eb-9fb5-05b0a111440d.mp4

Small parts:

https://user-images.githubusercontent.com/53969222/120491072-3b86fb00-c3b9-11eb-9855-9756002091c8.mp4

Models:
[models.zip](https://github.com/prusa3d/PrusaSlicer/files/6584934/models.zip)

bubnikv, 2021-06-02:
Duplicate of #6253 ?
This issue will be fixed in PrusaSlicer 2.4, the fix is already on master.

zakutin, 2021-06-02:
This is exactly the same issue number :) I just commented under the closed issue.

commit 08ca5b29 (2021-04-08): Fix slow cancellation of rasterization step

fixes #6253