prusa3d/PrusaSlicer #6255: Interface support layer still not generating with last changes

ProTubeVR, 2021-03-22:
### Version
_Version of PrusaSlicer used goes here_

Build from Master on 03/22/2021 especially to test out last support generation rework.
version_2.4.0-alpha0-34-gf5d386684

### Operating system type + version
Win 10 64x

### 3D printer brand / version + firmware version (if known)
Slic3r only Behavior

### Behavior
Several problems on support generation.
The top and bottom Z distance value look inverted,
the interface layers are generated only if the part's overhang is over 60° (none interface layer generated bellow this, i think i tried any possible setting without success)
[image]

Look like the problem wasn't solved for me

_Is this a new feature request?_

No

bubnikv, 2021-03-22:
> The top and bottom Z distance value look inverted,

Do you have a proof? Example data?

> the interface layers are generated only if the part's overhang is over
60°

That is on purpose. You don't need the dense supports on steep overhangs.
Sparse infill is sufficient for stability.

ProTubeVR, 2021-03-23:
Case 1 :
[image]
[image]

Interface layer sit at 3.30mm

[image]

The supported floor is printed at 3.65mm with 0.15mm layer height, so it's 0.20mm effective distance instead of 0.1mm

Case 2 :
[image]
No interface layer generated 
[image]
[image]

Case 3 : (0.1mm for top and bottom)
[image]
In this case interface layer sit at 3.40mm, and supported floor at 3.65mm, so 0.1mm real distance, so all is good in this case
[image]
[image]

On 60° limitation for interface layers,
I just don't get why you limit those, this should follow the support structure.
Making Support without interface layers is just an error to me, 
depending on the material used, I real need to set this value myself 
(right now my only way to get nice enough part with supports is to model the supports myself just due to this 60° limit, and it's SUPER annoying to know it's like this on purpose..... I just don't get why, so from user perspective it's a pure bug)

ProTubeVR, 2021-03-23:
The particular part i try to manufacture real need those interface layers from ~45° overhangs,
Tested with slic3r, Cura and simplify3D,
Only Slicer don't generate those interface layers, and the part looks bad on the areas without interface, and near perfect when i design myself the support structure,

So Sparse infill isn't sufficient at all.

bubnikv, 2021-03-24:
Please very please very much much please, would you please very much please provide your 3MFs where you can reproduce the issue please?

ProTubeVR, 2021-04-01:
> 
> 
> Please very please very much much please, would you please very much please provide your 3MFs where you can reproduce the issue please?

Hi,
sorry for late reply,
Sorry but can't share those 3MF for NDA reasons,
The problems described appear as soon you need to use supports structure.

For the 60% overhang threshold on interface layers :
I could find the param and nuke it in the source code, it's way BETTER

closed 2021-10-06 by bubnikv