prusa3d/PrusaSlicer #6256: First layer height impact speed on every other layers

eiffel31, 2021-03-22:
### Version
2.3.0

### Operating system type + version
On Linux

### Behavior

Depending on the height of the first layer, the printing speed of all other layers is changed.
The change can be moderate or very big.

First layer height -> total print time
0.18 -> 1h03'
0.19 -> 1h03'
0.20 -> 1h35' ???
0.21 -> 1h55' ???
0.22 -> 1h09'
0.23 -> 1h08'
0.24 -> 1h04'
0.25 -> 2h23' ???
0.26 -> 1h09'
0.27 -> 1h06'
0.28 -> 1h05'
0.29 -> 1h03'
0.30 -> 1h03'

This behavior is totally unexpected!

#### Project File (.3MF) where problem occurs
[accroche_bed.zip](https://github.com/prusa3d/PrusaSlicer/files/6185304/accroche_bed.zip)

rtyr, 2021-03-22:
For our info. All speeds are set to auto (legacy autospeed).

aknezevi-work, 2025-05-07:
I have a very similar, if not the same, issue.
I am using Variable Layer Height feature, which is very powerful, and first layer height is kind of a variable layer height.
However, having set speeds hard coded as mm/s when using Variable Layer Height is bad because print will either go too fast on some places or too slow when it could be printing faster.
Therefore, I want to use auto speed.
It is just not consistent with speed giving. It will unnecessarily slow down the print speed sometimes, depends really on the Variable Layer Height and the amount of times i press "smooth".
It might have to do with gap filling, but is more likely, at least by what I saw, to be an issue with Arachne perimeter generator, which is also super great feature to have.
I have several projects, sometimes Variable Layer Height "solves" the issue and the print is faster, sometimes it makes the issue and the print is slower. Sometimes it makes no difference. Depends on the geometry, when it really should not so much.

[Auto_Speed_3.zip](https://github.com/user-attachments/files/19976094/Auto_Speed_3.zip)
[Auto_Speed_4.zip](https://github.com/user-attachments/files/19976093/Auto_Speed_4.zip)

[Auto_Speed.zip](https://github.com/user-attachments/files/19976106/Auto_Speed.zip)

closed 2026-09-29 by prusaslicer-bot[bot]