prusa3d/PrusaSlicer #6257: PrusaSlicer generating unexpected top layers after slicing

tslater2006, 2021-03-22:
### Version
2.3.0-alpha1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 5, Marlin Firmware

### Behavior
Unexpected top layers are generated during slicing
To reproduce the issue, import the model found here: https://www.thingiverse.com/thing:2502903. It will import and render correctly:
[image]

After slicing:
[image]

Prusa reports no model issues, and Meshmixer inspector finds no problems either. 

There are many other comments on this file about this exact issue with Prusa https://www.thingiverse.com/thing:2502903/comments#comment-2903654

I've confirmed the model slices correctly in Cura, generating no top layers like PrusaSlicer is.

#### Project File (.3MF) where problem occurs
[Vallejo_Paint_Holder-3mf-project.zip](https://github.com/prusa3d/PrusaSlicer/files/6185413/Vallejo_Paint_Holder-3mf-project.zip)

tslater2006, 2021-03-22:
Worth noting that if you perform a cut on the model, and take of 0.01mm from the top, it slices correctly.
[image]

rtyr, 2021-03-22:
Already fixed on master (9c5dd35), the fix will be included in the next PrusaSlicer release. It also slices ok with 2.3.0 after repairing the model (right click on the model -> Fix through the Netfabb). Closing.

> Fixing model import of models with nearly degenerate triangles, which
> degenerate after transformation (translation or rotation).
> 
> There was certainly a bug in PrusaSlicer, however this bug was triggered
> mostly by the ASCII STL files generated by OpenSCAD, likely due to insufficient
> number of decimal digits when saving an ASCII STL, STL export should contain single precision floats
>

closed 2021-03-22 by rtyr