prusa3d/PrusaSlicer #6262: Feature Request: Include calibration / test objects

ulab, 2021-03-24:
### Version
Doesn't matter / 2.3.0+

### Operating system type + version
Doesn't matter / macOS 11.2.3 (20D91)

### 3D printer brand / version + firmware version (if known)
Doesn't matter / Original Prusa i3 MK3S+ MMU2S

### Behavior

This is a new feature request.

It would be nice to have some calibration / test objects readily available in PrusaSlicer. Things like:

  * calibration cubes
  * ringing tests
  * temperature towers for different materials
  * retraction tests (stringing)
  * bridging tests
  * Benchy(!) ;)
  * MMU filament swap test cubes

I keep having to go to the 3d object websites and search for the any of these every now and then. Having a small repository of defined objects in PrusaSlicer would help. It would also help with comparing results with others, since everybody can use the same objects.

neophyl, 2021-03-24:
Duplicate of #6168

ulab, 2021-03-24:
Thanks. Had not searched for the proper phrases it seems.

closed 2021-03-24 by rtyr