prusa3d/PrusaSlicer #6264: Long preset names hide "(modified)" tag in Print settings dropdown

murk-sy, 2021-03-25:
### Version
2.3.0+win64

### Operating system type + version
Windows 10

### Behavior
When print settings name is too long, the "(modified)" tag is not visible. This also applies for Filament and Printer. This can cause an user preparing multiple prints in a row to not reset the settings after making minor modifications or when loading older 3mf files.
Example:
[image]

#### Expected behavior
When the print settings, filament or printer name is too long and is modified, either:
- Truncate the name with elipses so the tag is visible - e.g. "0.2mm Quality @Mini Structural pa... (modified)"
- Add right justified icon denoting the settings have been changed (may be less visible than "(modified)" for users with short profile names)
- **Colorize modified settings name orange to be consistent with other settings (likely easiest to implement and consistent with other parts of the slicer)**

_Is this a new feature request?_
Minor enhancement

Jaxx005, 2021-12-24:
I also think this would be a useful feature. Users can have presets for all sorts of reasons and longer-filenames are (imo) vital to printing models in a reproducable/consistent manner (ie. by keeping groups of settings intact/seperate).

_SUGGESTIONS_:  
Make text box width user-adjustable (eg drag right-hand edge)? 
This might be more acceptable than just making the text box wider?

Either way, it would show all but the exceptionally long filenames. 

Similarly, the pull-down list of Filenames (and/or their ToolTip popups) would ideally show the whole filename of each preset.

All this probably applies to Open/Save/Load operations.

As @murk-sy mentioned, perhaps indicate "modified" with an icon, as elsewhere ["modified" is never actually part of the filename anyway is it?  ;)  ].    Or just _prefix_ the filename with an asterisk (*) instead?

VidiotGeek, 2023-08-24:
Since this is still open--yes to any or all of the above. Tooltip, color change, resizable dropdown menu and or dropdown menu that dynamically adjusts to longer preset names, even if only to a certain limit.

murk-sy, 2026-09-24:
Resolved in 3.0.0-alpha12

Unsaved changes are now marked with purple consistently over the UI

[image]

closed 2026-09-24 by murk-sy