prusa3d/PrusaSlicer #6265: Wipe tower size for direct drive independent extruders

pcsentinel, 2021-03-25:
### Version
2.3.0

Is there any way, or if not can there be a setting for wipe tower size. I know you can change a single dimension, but it appears as though it recalculates to obtain the same area. The issue is that if you are using two direct drive extruders the the wipe tower just uses excessive amounts of filament.

lukasmatena, 2021-03-25:
You can set the amount that should be purged for each filament pair:

[image]

pcsentinel, 2021-03-25:
Thanks, that works.

Regards

Paul Clarke

[2021-03-25-165644_3840x1080_scrot]<https://user-images.githubusercontent.com/33929324/112503710-72261100-8d8b-11eb-9b12-f9491dd57634.jpg>

closed 2021-03-25 by lukasmatena