prusa3d/PrusaSlicer #6267: Uninstalling PrusaSlicer 2.3.0 does not clear all related Registry entries

OrangeHedgehog, 2021-03-25:
### Version
2.3.0
### Operating system type + version
Windows 10 Pro (20H2)

### Behavior
* When uninstalling PrusaSlicer 2.3.0 registry entries remain.
* Uninstall PrusaSlicer. This may be achieved through using the original installer file, or through WIN Key + I \Apps\Apps and Features.
* _Expected Results_
All registry entries removed
* _Actual Results_
Registry entries remain.  When PrusaSlicer 2.3.0 is reinstalled (following computer restart) it retains the list of Recent files and profiles.

_Is this a new feature request?_
Yes

Related issue: Fix missing Recent files list, Windows 10 Start Menu

OrangeHedgehog, 2021-03-25:
I should also add, I tried the Repair option available through running the 2.3.0 Win standalone installer, it didn't help.

fe60, 2021-03-29:
The list of recent projects is retrieved from the `PrusaSlicer.ini` in folder `%USERPROFILE%\AppData\Roaming\PrusaSlicer`.
Even though an installer should remove all stuff related to an application, I appreciate the configuration is not removed when PrusaSlicer gets uninstalled. That way I have uninstalled old versions and installed a new ones without losing my settings.

So if this gets cleaned during the removal of PrusaSlicer, I would like to get an option to let the user decide whether the configuration in `%USERPROFILE%\AppData\Roaming\PrusaSlicer` is removed or not.

Janys, 2021-04-19:
Hello, files in %USERPROFILE%\AppData\Roaming\PrusaSlicer are intentionally not removed (e.g. for the purpose of migration to a newer PrusaSlicer etc.) 

As far as I am aware, the installer can only delete files from the following registry path:
HKLM\Software\Prusa Research\PrusaSlicer

Can you please give me an example of which registry entry is not deleted?

OrangeHedgehog, 2021-05-27:
> Can you please give me an example of which registry entry is not deleted?

Sorry, I have to admit I was making an assumption about the cause being a registry entry, simply because that would seem to make the most sense. 

I'll try to have another go at uninstalling and re-installing.  I'll check the Registry path provided prior to uninstalling, after the uninstall, then again after the reinstallation.

I'll also check the .ini file in the folder fe60 mentioned.

OrangeHedgehog, 2021-05-29:
Okay, update.  I deleted the PrusaSlicer folder within the `%USERPROFILE%\AppData\Roaming\` path between uninstalling the previous version (having made a copy), restarted the computer, then installed ver 2.3.1 from the Prusa Website*

Once the application was installed again I reinstated some of the .ini files in the `Roaming\PrusaSlicer` folder and previously open files were once again showing on the right-click.  I think the learning point/**opportunity to improve the poduct** here is that there should be an **option** to remove all files when uninstalling PrusaSlicer.**  For surity, this option **_should be combined_** with the option to export/backup the Print, Filament and Printer settings (presumably by utilising the Export Config modules) as part of the uninstall activities.  This would serve as a reminder, not force, but enable the export of the configurations at such a time.  It could be accompanied with suitable descriptive wording.  It might also be an option to consider including in the Repair feature?

Sadly, I don't know if we can be sure what _exactly_ went wrong and ultimately the solution was to uninstall, restart, reinstall. 

*Side note: I found out the Win 64 version in GitHub is not the proper installer.  I suppose that's intended (?), but it was not helpful to me and resulted in it taking longer to resolve the issue because the GitHub version didn't 'install' the software so Windows 10 could add it to the Start menu.

closed 2026-09-29 by prusaslicer-bot[bot]