prusa3d/PrusaSlicer #6269: Dual Extrusion Non Print Moves

pcsentinel, 2021-03-25:
### Version
3.2.0

It appears that the initial move to the wipe tower does not respect the Non print moves setting. In the attached, the non print move is 160mm/s but there is no G1 FXXX before the wipe tower move so the move is performed at default printer speed.
[a2e3ad70-ca18-47a1-9709-662c2f96b08d.zip](https://github.com/prusa3d/PrusaSlicer/files/6206783/a2e3ad70-ca18-47a1-9709-662c2f96b08d.zip)

rtyr, 2021-03-29:
I don't see any issue. The last G1 command before that travel is with F9600 (160mm/s), therefore the travel to the wipe tower will also use F9600. Or does the RepRapFirmware behave differently?

[image]

pcsentinel, 2021-03-29:
HI, but if you look at the code around the first tool change you will see that the move of Tool1 to the wipe tower location is doen at 2400.:

G1 X99.750 Y27.750 F9600.000 ; Travel to a Wipe Tower
G11 ; unretract
;HEIGHT:0.200000
;TYPE:Wipe tower
;WIDTH:0.500000
;--------------------
; CP TOOLCHANGE START
; toolchange #1
; material : PET -> PET
;--------------------
M220 S100
; CP TOOLCHANGE UNLOAD
;WIDTH:0.500000
G1  X99.250 Y27.500  
M572 D0 S0
;WIDTH:0.500000
G1 F2000
G1  Y27.000  F2400
G4 S0
; Filament-specific end gcode 
;END gcode for filament

T1 ; change extruder

G1  X99.250 Y27.000  
; Filament gcode

G4 S0
G1  Y27.500 
; CP TOOLCHANGE WIPE
G1  X90.250  E0.4036 F800

rtyr, 2021-03-29:
Initial travel from the object to the wipe tower is done with F9600. Only those tiny travels inside the wipe tower are done with fixed feedrate of F2400. I still don't see any issue here, what problem does it cause?

rtyr, 2021-03-29:
Oh now I understand, I somehow supposed you are using single nozzle multimaterial, but you are not.

rtyr, 2021-03-29:
It should be hopefully fixed along with https://github.com/prusa3d/PrusaSlicer/issues/5483.

closed 2026-09-29 by prusaslicer-bot[bot]