prusa3d/PrusaSlicer #6270: Segfault during startup in prusa-slicer-git 2.3.0.r24.gd06aa6069-1

jernst, 2021-03-25:
Arch AUR version on Gnome / x86_64 from here: https://aur.archlinux.org/packages/prusa-slicer-git/

During startup:

```
./include/wx/dynarray.h(837): assert "uiIndex < m_nCount" failed in Item().
```

Shown in the GUI. Then segmentation fault. Cannot run.

NekoNoor, 2021-03-26:
the version you listed in the title is the AUR version number, not the version it actually compiles `-git` AUR packages will compile whatever is latest on the main branch of the git repository. in your case that would probably be `13c178b7f788b548d48daf894771c1ce48ff0bd1`

i figured out this bug was introduced in commit `4496e2a8ce2371edfa9153e4f879b3b1e94a3b65`, the last working one is `991fa67fd1e71dad6d2d5d10fc6e98ba714f8a9f`
to compile a working build from the pkgbuild, you have to edit the `prepare()` section of the pkgbuild to include `git checkout 991fa67fd1e71dad6d2d5d10fc6e98ba714f8a9f` in between the `cd` and `patch` commands

that will compile and install prusa-slicer-git-2.4.0.alpha0.r41.g991fa67fd-1 which seems to work

inanimateobjectsaretryingtokillme, 2021-04-15:
The segfault happens when prusa-slicer is built with DSLIC3R_WX_STABLE=ON (wxgtk3 3.0.5.1); seems to run fine with wxgtk3-dev (3.1.4) instead / DSLIC3R_WX_STABLE=OFF.

- confirmed last working: 991fa67fd1e71dad6d2d5d10fc6e98ba714f8a9f
- confirmed first segfault: 4496e2a8ce2371edfa9153e4f879b3b1e94a3b65
- confirmed segfaul still happens with: eb428d48fc41c78367cbfad16165ab1138f3527d

```
[2021-04-15 16:38:00.234494] [0x00007f1b49d7a080] [trace]   System language detected (user locales and such): en_US
[2021-04-15 16:38:00.404284] [0x00007f1b49d7a080] [trace]   Best translation language detected (may be different from user locales): en_US
[2021-04-15 16:38:00.404315] [0x00007f1b49d7a080] [trace]   Switching wxLocales to en_US
[2021-04-15 16:38:00.541009] [0x00007f1b49d7a080] [info]    Checking if indices need to be installed from resources...
[2021-04-15 16:38:00.682525] [0x00007f1b49d7a080] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 112MB; Shared memory: 73MB; Private memory: 38MB; Peak memory usage: 112MB
[2021-04-15 16:38:00.683768] [0x00007f1b358f1640] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object8647195901576075186 registered. Starting listening for messages.
./include/wx/dynarray.h(837): assert "uiIndex < m_nCount" failed in Item().

(prusa-slicer:2091045): Gtk-WARNING **: 16:38:00.709: gtk_window_set_titlebar() called on a realized window
[1]    2091045 segmentation fault (core dumped)  prusa-slicer --loglevel 6
```

For PKBUILDS, if you add the commit to the url, the resulting package will be named correctly. 
`url='https://github.com/prusa3d/PrusaSlicer#commit=991fa67fd1e71dad6d2d5d10fc6e98ba714f8a9f'
`
> warning: downgrading package prusa-slicer-git (2.4.0.alpha0.r42.g4496e2a8c-1 => 2.4.0.alpha0.r41.g991fa67fd-1)

(Building, running and slicing worked for me with wxgtk dev + 48a93e40f and several other versions I tried, so this might be partially mixed up in or work around #6345 etc, didn't check any further though.)

NekoNoor, 2021-04-15:
this did not seem to work for me, i installed wxgtk3-dev from the aur and changed the compile variables to DSLIC3R_WX_STABLE=OFF. when launching i get several wxgtk errors and it still doesn't launch

clytle374, 2021-05-07:
I've got, what I believe, to be the same fault.  And I might be able to add to this a little.
Running latest git.  Problem started some weeks ago.  Running Gentoo Linux.  Here is the output I get when trying to start Prusa-slicer

`./prusa-slicer
[2021-05-07 03:58:52.812926] [0x00007f41617bf300] [trace]   Initializing StaticPrintConfigs
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r2/work/wxWidgets-3.0.4/include/wx/dynarray.h(837): assert "uiIndex < m_nCount" failed in Item().

Trace/breakpoint trap
`
Here is the interesting part.  As gentoo is build completely build from source code, wxwidgets is built in a temp dir.  If you look in the failure the problem is that it's pointing to dynarray.h this file does not exist in this location.  I'm sure of this because it's actually a ram drive and therefor deleted on reboot.  I asked around the gentoo forums what is wrong with the wxwidgets(wxGTK) that prusa config is getting this incorrect location.  They think it's the method prusa is using to find it.  I have to run wx stable for now.
Hope this provides some insight.
Cory

Tony763, 2021-05-11:
Hi,
 If I build without deps, only with system installed. I get `[0x00007febe2b83e80] [trace]   Initializing StaticPrintConfigs` `(SIGSEGV) (core dumped`.

When I try to build deps:
```
PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/cairo.cpp:24:10: fatal error: cairo.h: Adresář nebo soubor neexistuje
   24 | #include <cairo.h>
      |          ^~~~~~~~~
```
But `libcairo2-dev` is already newest:
```
pkg-config --modversion cairo
1.16.0
```

YuSanka, 2021-05-11:
Thanks for your reports. @jernst , @NekoNoor , @inanimateobjectsaretryingtokillme and others, can I ask to test [this ](https://github.com/prusa3d/PrusaSlicer/commit/de1d36cc9aa3890208b2e9bfcc30606972539959) commit?

clytle374, 2021-05-12:
> Thanks for your reports. @jernst , @NekoNoor , @inanimateobjectsaretryingtokillme and others, can I ask to test [this ](https://github.com/prusa3d/PrusaSlicer/commit/de1d36cc9aa3890208b2e9bfcc30606972539959) commit?

This fixed my issue.
Thanks
Cory

Tony763, 2021-05-15:
Hello, for me not. 
With latest master, compiling deps still end with error about missing `<cairo.h>`.

When I try compile without deps with systems only, I now get error about missing `DBUS` and `TBB_LIBRARIES`. 
Its odd as before update to latest commit, I was able at least to build it with seg fault from above.

```
$ cmake -DSLIC3R_STATIC=1 -DSLIC3R_GTK=3 ..
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 3
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: 1
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: 1
-- CMAKE_PREFIX_PATH: (default)
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (DBUS) does
  not match the name of the calling package (DBus).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindDBus.cmake:59 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:202 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Boost::boost exists
CMake Error at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_LIBRARIES) (found version "2020.3")
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake:286 (find_package_handle_standard_args)
  CMakeLists.txt:374 (find_package)

-- Configuring incomplete, errors occurred!

```

bubnikv, 2021-07-30:
I believe the original issue has been fixed with https://github.com/prusa3d/PrusaSlicer/commit/de1d36cc9aa3890208b2e9bfcc30606972539959
Closing

closed 2021-07-30 by bubnikv