prusa3d/PrusaSlicer #6271: Feature request: "Plater + Slice" Tab

LizRidley, 2021-03-25:
### Version
2.3.0+

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
When making various changes on Print and Filament settings tabs I find going to the Plater tab at the top of the screen and then moving down to the slice button at the bottom a bit long winded.

I was just thinking that if there were an extra tab "Plater + Slice" tab at the top it would save a step in the process.  I'm relatively new to 3D printing so I'm often changing infill types, perimeters and such before deciding on final settings.

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
N/A

combolek, 2021-03-25:
From anywhere in the GUI you can use Ctrl-R to slice and Ctrl-1 to switch to the Plater tab (in either order).

LizRidley, 2021-03-26:
Excellent, I wondered if I'd been missing a trick, thanks very much.

closed 2021-03-26 by LizRidley