prusa3d/PrusaSlicer #6273: Seam Gap Distance

TheRunnerPi, 2021-03-26:
Hi,

is there a way to change the seam gap distance, as seen on the Screenshot?
I would like to lower the gap, because my printer prints these gaps on the physical print.
Is there maybe a Setting in the config somewhere?

[image]

Thanks,
Daniel

rtyr, 2021-03-26:
There are no gaps in the g-code, it is just a visualization of a seam. Seams are indeed visible on the physical prints. How much depends on your printer and settings. Closing.

rtyr, 2021-03-26:
Sry, my initial answer was not correct. There is actually some tiny gap to better hide a seam on closed loop, calculated `0.15 * nozzle diameter`, it is hardcoded. It shouldn't cause actuall gaps in the prints though, they are likely caused by hw/settings. Seams are visible on physical prints.

TheRunnerPi, 2021-03-26:
Thanks for the quick answer rtyr,

I dont understand why a normal retract in mid print is working, but the retract at the Seam Gap messes all up.

Here are some Pictures of the print..

Seam straight:
[image]
Seam random:
[image]

rtyr, 2021-03-26:
It looks like underextrusion/insufficient pressure at the beginning of the loop (after retraction/travel). It is worse when the seams are aligned because you are printing over that underextruded perimeter again on the next layer. 

It is about settings and hw. You should check your retraction settings, linear advance K value (if used). You can also experiment with "extra lenght on restart" setting.

TheRunnerPi, 2021-03-27:
> 
> 
>... it is calculated `0.15 * nozzle diameter`...

With a 0.4 nozzle thats 0,06mm, if i choose that as "extra lenght on restart" the seams are fine.

Thanks and close

closed 2021-03-27 by TheRunnerPi