prusa3d/PrusaSlicer #6274: Problems in the spiral vase mode

Hurin24, 2021-03-26:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Creality Ender 3

### Behavior
The problem is the wrong slicing of the model in the slicer, when I use the spiral vase mode, then after the end of filling the bottom, at the beginning of printing the spiral, there is a thickening. This happens on all the models I've tried.
To repeat this problem it is enough to slice a regular cube 20 by 20mm in the spiral vase mode.
I also checked the Code generated by PrusaSlicer, and found there is not a proportional supply of plastic depending on the height of the layer. At the beginning of the spiral printing, the slicer instructs the printer to squeeze out a volume of plastic for a layer with a height of 0.4, but there is a smaller distance between the nozzle and the bottom of the model, and because of this, the plastic spreads out to the sides.
I would like you to make a proportional flow of plastic depending on the height on the first layer of the spiral mode.

Black plastic - PLA, white - SBS

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Here you can see how at the first layer of the vase, after filling the bottom, the spiral printing begins, and regardless of the height of the nozzle from the bottom of the model, the extruder moves by 2.28029 mm

[image]

The same thing happened in PrusaSlicer 2.2

Good luck

rtyr, 2021-03-26:
Same as https://github.com/prusa3d/PrusaSlicer/issues/2795, https://github.com/prusa3d/PrusaSlicer/issues/3712, etc.

Linear tapering of extrusion rate at the 1st spiral vase layer is already implemented in 2.3.0, but it works only for relative extrusion distances (you are using absolute extrusion distances).

See https://github.com/prusa3d/PrusaSlicer/issues/2795#issuecomment-741864142.

Hurin24, 2021-03-26:
> Same as #2795, #3712, etc.
> 
> Linear tapering of extrusion rate at the 1st spiral vase layer is already implemented in 2.3.0, but it works only for relative extrusion distances (you are using absolute extrusion distances).
> 
> See [#2795 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/2795#issuecomment-741864142).

not work
[image]

rtyr, 2021-03-26:
As I mentioned, it works only with relative extrusion distances, you are using absolute extrusion distances.

[image]

Hurin24, 2021-03-26:
> As I mentioned, it works only with relative extrusion distances, you are using absolute extrusion distances.
> 
> [image]

Oh, I didn't know about the existence of this tick, you solved my two-week problem, thank you very much

[image]

[image]

Hurin24, 2021-07-29:
The problem has returned, despite the fact that I use relative extrusion, the first layer of the spiral protrudes, version 2.3.0. All that has changed since I solved the problem is that I put the Clipper on the printer board.
But I don't think that's the problem with it. Could there be a problem with the fact that I use a 1.6 mm nozzle? I use a layer height of 0.6 mm and a line width of 268% of the layer height, which is approximately 1.6 mm. It is very strange that the problem has returned...

closed 2026-09-29 by prusaslicer-bot[bot]