prusa3d/PrusaSlicer #6275: Slicer splits internal infill (100%) and solid infill, could have been replaced by a continuous layer

cnrd, 2021-03-26:
### Version
2.3.0

### Operating system type + version
macOS 10.15.6

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* _Describe the problem_
When slicing a model with 100% infill, the slicer will create internal infill and solid infill separately in cases where it would make more sense to combine these.
* _Steps needed to reproduce the problem_
Create a model where there is a partial cutout, set the infill to 100% rectilinear
* _Expected Results_
The slicer should not split out solid infill and top solid infill, when it does not make sense to do so.
* _Actual Results_
The slicer splits out solid infill and top solid infill from internal infill, even when it makes sense to just make one continuous layer.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
    [image]
    [image]
    [image]

_Is this a new feature request?_
Unknown

#### Project File (.3MF) where problem occurs
[foot v10.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6213361/foot.v10.3mf.zip)

wavexx, 2021-06-18:
I think this just needs to set more defaults when switching to 100% infill.
When switching to 100% infill, in addition to the current proposed changes, it also needs to:

- Disable "Ensure vertical shell thickness", since this will create discontinuities on the perimeters.
- Set top/bottom solid layers to 1.

If you apply those settings the volume shouldn't be split.
Could you try?

timur-tabi, 2026-09-02:
I would still like to see this bug fixed.