prusa3d/PrusaSlicer #6276: Elephants Foot Compensation Feature Request

retroglyphic, 2021-03-26:
### Version
2.3.0+

### Operating system type + version
macOS Sierra V10.12.6

### 3D printer brand / version + firmware version (if known)
several.. Blue Eagle Labs Delta, Ender 3, DiY machines

### Behavior
occasional Elephants Foot Problem

--------------------------------------------------------------------

I tend to use PrusaSlicer as my goto slicer. Thank you for all the work and making an interface that makes sense to me. I used to use Cura (and it too is good), but the interface has become less user-friendly (for me.)

Regardless of the slicer I use, occasionally I have certain models, with some filaments, where I get rather bad Elephants Foot. Even when I use the PrusaSlicer Elephants Foot compensation setting, sometimes the Elephants Foot will still extend several layers up into the print.

Toward the goal of removing it, I wrote a little SCAD ( https://www.thingiverse.com/thing:4800868 ). This allows a user to select a model, select the number of affected layers, first layer height, layer height and amount of compensation needed. It will then use openSCADs projection and offset tools to "slice", offset and re-extrude the first N number of layers. Basically, it creates a custom chamfer to parametrically compensate for Elephants Foot. It works OK, in many situations. However, the biggest issue with using SCAD offset for this is that there is always potential for overly removing areas of thin details. How else to say... one could easily offset to the point where thin lines no longer exist.

It seems that PrusaSlicer might use an algorithm that can preserve a minimum line width.

So, with the goal of trying to leverage PrusaSlicer, I am looking at ElephantsFootCompensation.cpp This is my first perusal. In simple terms, I'm trying to work out how the slicer gets the path from ElephantsFootCompensation.cpp

My goal is this: What I think would be awesome would be if "Elephants Foot Compensation" was not just a first layer compensation, but an N layer compensation. So, you could print a cube without compensation, then measure. If there is Elephants Foot, you'd enter not only the amount to offset, but the number of layers (or alternatively, the affected amount in mm)

If I can work out how the slicer gets the path from ElephantsFootCompensation.cpp and if I can figure out how ElephantsFootCompensation.cpp gets the amount to offset, then I can try to implement.

There are many who think that the only cause for elephants foot is bad first layer height. And I absolutely agree that bad first layer height is a primary cause. However, I believe there may be other things to consider. And that being able to compensate for these is valid.

I can currently print a 20 x 20 x 20mm calibration cube at 0.2 layer height for all layers, with a number of different PLA filaments and the results from my printer produce a cube that varies slightly, but within 19.97 to 20.03 in all dimensions. It's not perfect, but fairly well calibrated and with most filaments, I get no elephants foot. If I do a single layer test of the same, that first layer feels smooth and measures very close to 0.2. So I don't feel like I have a bad first layer height in terms of dimensional accuracy.

However, if I try to print the same 20 x 20 x 20mm calibration cube at 0.2mm using a certain brand of carbon fiber infused PLA, for example, then I will get elephants foot - and not just on the first layer. That elephants foot can extend several layers up into the print. If I change the z offset or the first layer height, I can remove the elephants foot. However, then I have dimensional accuracy issues. It'll still be accurate in the X and Y. But the Z height will be off, predictably, by the amount that I changed the first layer. 

In the above example, this isn't a bed adhesion issue. This is a dimensional accuracy issue. So, rather than change the offset or first layer height to try to compensate for differences in filament performance, I propose to modify the first N number of paths to compensate. 

I propose that the interface include an additional argument in the "Print Settings" tab, directly under the "Elephant foot compensation" something like "Number of affected layers".  The first layer would get offset by the amount entered in "Elephant foot compensation". Subsequent layers would get offset less and less up to layer N+1 which would then not be offset.

Thank you for all the hard work

Steve

pylorak, 2022-07-12:
Seconded.

In my experience, for prints that have an elephant foot at all, the elephant foot often affects multiple layers. An elephant foot can have multiple reasons (not just bad first layer height). 

Shorter version of the post above:
The slicer should provide a second parameter in addition to the current elephant foot compensation value. The new parameter would determine how many layers the compensation should affect, and gradually taper out the compensation over the given number of layers.

Example:
So if elephant foot compensation (currently available parameter) is set to 0.4mm, and the number of compensated layers (proposed new parameter) is set to 4, the 1st layer would be compensated by 0.4, the 2nd by 0.3, the 3rd by 0.2, and the 4th by 0.1mm.

pylorak, 2022-07-14:
Seems PR #6813 implements this, but is not merged yet.

stepikovo, 2022-07-17:
I think this FR can be closed as a duplicate as mentioned by @pylorak

closed 2022-07-17 by stepikovo