prusa3d/PrusaSlicer #6280: Feature Request: Custom G-code section for retraction and derectraction

yru2501, 2021-03-27:
### 2.3.0

Would be great to include variables such as "travel after retraction" and "last print move feedrate".

rtyr, 2021-03-31:
Why? What is the use case?

yru2501, 2021-03-31:
Starting and ending of extrusion is crucial. This would give way to have even more control over it. Imagine dividing it into releasing mechanical "spring" tension and adding pressure from feed-rate component and then constant "slurp in" and not going crazy when next move is relatively short. Some fancy string break moves and so on. Decoupling in spot retraction from z-hop and wipe. I was experimenting with it for layer change with projects like airless tires for mars rover. Having this power over every retraction would be almost like skeinforge was back :) how about individual extrusion multipliers for each type of print move while we are at it :)

closed 2026-09-29 by prusaslicer-bot[bot]