prusa3d/PrusaSlicer #6281: Feature Request: option to rebind camera pan to middle mouse button.

RedshiftVelocities, 2021-03-27:
I would appreciate an option to be able to pan the camera using the middle mouse button (similar to most CAD software) so that the buttons to select an object and to pan the camera are different. Thanks!

rtyr, 2021-03-31:
What do you mean? You can pan the camera using the middle mouse button.

neophyl, 2021-03-31:
I think they mean the left mouse button.  At the moment its left button to select an object but also to orbit the plater.  While the middle mouse button and the right mouse button both pan the camera around (I hope I got those the right way round).  

If you had the middle mouse button doing orbit and the right mouse button for pan you could reserve the left purely for selection.   That way you wouldn't accidentally grab an object when you just want to move your view port.

RedshiftVelocities, 2021-05-09:
> If you had the middle mouse button doing orbit and the right mouse button for pan you could reserve the left purely for selection. That way you wouldn't accidentally grab an object when you just want to move your view port.

Sorry that I'm waaay late in this, but this is exactly correct. The intent is that each mouse button always does exactly one thing; the left button always selects something, the middle button always orbits the camera, and the right button always gives additional options in a drop-down menu. 

At minimum, a massive improvement to usability would be to bind camera orbit to the right button. That way, dragging the camera and dragging a part use separate buttons and thus will never get mixed up.

All of this is just my two cents, to be fair, but they're small features that would vastly improve my enjoyment of the software.

closed 2026-09-29 by prusaslicer-bot[bot]