prusa3d/PrusaSlicer #6284: Crash @ connecting to Octoprint

SanderRave, 2021-03-29:
### Version 2.3.0

### OSX 11.2.3

### OctoPrint 1.5.3

### Behavior
* Reconnect to the OctoPrint Server after it obtained a new IP address
* In slicer overview, top bar, click Printer presets --> gear symbol 
* Adjust URL / IP address, test
* CRASH
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ Yes

[20210329 OSX log PrusaSlicer Crash.txt](https://github.com/prusa3d/PrusaSlicer/files/6221871/20210329.OSX.log.PrusaSlicer.Crash.txt)

gordyj2000, 2021-04-18:
I'm having this same issue too. Version 2.3.1rc1 for Apple Silicon, Octoprint 1.5.3, and OS X 11.2.3. It either locks up or doesn't give me the button to send gcode to Octoprint.

bubnikv, 2021-04-19:
Is it reproducible? Do you get it every time?

@YuSanka 
```
Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.prusa3d.slic3r/           	0x00000001087ba4ff Slic3r::GUI::UnsavedChangesDialog::build(Slic3r::Preset::Type, Slic3r::PresetCollection*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, wxString const&) + 1951
1   com.prusa3d.slic3r/           	0x00000001087bbe2a Slic3r::GUI::UnsavedChangesDialog::UnsavedChangesDialog(Slic3r::Preset::Type, Slic3r::PresetCollection*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 490
2   com.prusa3d.slic3r/           	0x0000000108650b5f Slic3r::GUI::Tab::may_discard_current_dirty_preset(Slic3r::PresetCollection*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 63
3   com.prusa3d.slic3r/           	0x000000010864f8ae Slic3r::GUI::Tab::select_preset(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 750
4   com.prusa3d.slic3r/           	0x00000001085ca3c1 Slic3r::GUI::PhysicalPrinterDialog::OnOK(wxEvent&) + 3809
```

SanderRave, 2021-04-19:
For me it was reproducible. Currently my octoprint is offline

YuSanka, 2021-04-20:
Thanks for your report. This bug is allready fixed with 2f23d5af9d6b5dd670dfb7d06eedbe0e6d2960c6 and will be a part of PrusaSlicer 2.4.0

closed 2021-04-20 by YuSanka

SanderRave, 2021-04-20:
Great!
Keep up the good work, highly appreciated.
Best regards,

Sander Rave
Oude Delft 121, 2611 BE, Delft [NL]
+31626564889