prusa3d/PrusaSlicer #6286: Modifier not working as expected

scrumpydc, 2021-03-29:
### Version
_Version of PrusaSlicer used goes here_
2.3.0
### Operating system type + version
windows 10 

### 3D printer brand / version + firmware version (if known)
marlin firmware haven't got that far yet 
### Behavior
the modifier is not doing infill only even with perimeters and different infill amounts just ignore's it how do I solve this 
[image]

rtyr, 2021-03-29:
Please povide your 3mf project file.

scrumpydc, 2021-04-03:
how do I

 do that it has no option for exporting 3mf and cannot attach an stl ?

scrumpydc, 2021-04-03:
[infill test.zip](https://github.com/prusa3d/PrusaSlicer/files/6252880/infill.test.zip)

neophyl, 2021-04-03:
I'm afraid the stl is not of any use on its own.  You reported an issue with the modifiers not working, as such the developers need to see how you have the modifiers applied and what settings you are using as well as the model.  That is why they asked you for a saved project file (.3mf).  
A .3mf file when created using Prusa Slicers File>Save Project As menu command will contain not only the model but ALL the settings you are using.  This is required to duplicate any issue you may be having.

Please zip up and attach the project file as requested.

scrumpydc, 2021-04-03:
ok i could not find info on how to do that thanks hear it is 
[infill test.zip](https://github.com/prusa3d/PrusaSlicer/files/6253063/infill.test.zip)

neophyl, 2021-04-03:
Is this what you were trying to achieve ?
[image]

I don't work for Prusa, just one of their software users :)  however this is down to the fact that you had  Print Settings>Infill> Only infill where needed ticked.  If you added 1 or more top layers the infill was generated as its needed to support the top layers.  With the top infill turned off then no infill is needed so it didn't slice the infill in your modifier.  As such the modifiers are working normally and as expected.

scrumpydc, 2021-04-03:
Thanks for the help did not see that XD could do with you on my discord for things like this I have only just started using Prusa slicer so still getting used to it came from another program because I like the paint on supports and some of the other nice features so thanks again

closed 2021-04-03 by scrumpydc