prusa3d/PrusaSlicer #6287: Prusa Slicer once startet on a other Display: Freeze

muellerbru, 2021-03-29:
### Version
Version 2.3.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,257MB
OpenGL installation
GL version:   4.6.0 NVIDIA 457.51
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 3070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
[... 349 lines cut ...]
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s /MMU2S, Use Octoprint (works fine) lates Firmware: prusa3d_fw_3_9_2_MK3_1_0_6_MMU2

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

Opend Prusaslicer on Screen 0 (2550x14440), moved window to Screen 2 (1920 x 1080), closed PrusaSlicer, tried to open PS again: slicer startet on Screen 2 and freeze only with Menu, but without Bed view. Waitet about 4 Minutes, nothing happened. Killed PS by Taskmanger, Found another, older Issue: PrusaSlic3r freezing at startup (Win 10) #2939 with more or less the same behaviour, but closed. Changed the Value in Prusa INI (window_mainframe, an then the Slicer started proberly.
[image]
[image]

Reproduceable:
OPen PS on Screen one, move window (fullscreen) to the other Display, open another PS instance on Screen one, close PS on Screen one (2556x1440), then on Screen two -> Screenshot and Ini as attached: PS freeze.

#### Project File (.3MF) where problem occurs
No Project open, Start PS without any Project.

fe60, 2021-03-31:
Similar to #5977

bubnikv, 2021-09-25:
May be similar to https://github.com/prusa3d/PrusaSlicer/issues/2939

closed 2026-09-29 by prusaslicer-bot[bot]